MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #3: unsuccess to unlock （system version：1.0.20.0.UNCCNXM）

nenemiku, 2023-11-23:
看到其他的B站评论有说1.0.14以后的版本解锁会失败。我这边的操作是成功连接后，点击解锁移动数据被关闭，提示解锁失败，CMD里显示
 [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in J:\mi\php-8.3.0-nts-Win32-vs16-x64\bypass.php:230
Stack trace:
#0 J:\mi\php-8.3.0-nts-Win32-vs16-x64\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in J:\mi\php-8.3.0-nts-Win32-vs16-x64\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in J:\mi\php-8.3.0-nts-Win32-vs16-x64\bypass.php:230
Stack trace:
#0 J:\mi\php-8.3.0-nts-Win32-vs16-x64\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in J:\mi\php-8.3.0-nts-Win32-vs16-x64\bypass.php on line 230

MlgmXyysd, 2023-11-24:
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass?tab=readme-ov-file#%EF%B8%8F-how-to-use

> Enable OpenSSL and Curl extension in php.ini.

chrisuhg, 2023-11-24:
指定你php的ext文件夹位置。
[image]

closed 2023-12-21 by MlgmXyysd

SVENDEMAN100, 2024-09-02:
This does NOT solve the issue unfortunately.

> 指定你php的ext文件夹位置。 [...]