MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #4: Uncaught TypeError: decryptData

alzubaidi, 2023-11-23:
Fatal error: Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in `Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php` on line 333 and defined in `Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php:228`
Stack trace:
#0 `Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php(333)`: decryptData(NULL)
#1 {main}
  thrown in `Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php` on line 228

MlgmXyysd, 2023-11-24:
Looks like logcat fetched null, try it again

Windforce17, 2023-11-24:
Try set `id` to empty string if you are using new adb version and only one device connected.
```
$id = $adb -> getDeviceId($device[1], true);
$id="";
```

alzubaidi, 2023-11-24:
Setting `$id` to empty string worked for me. However, it says
```
[2023-11-24] [15:06:50] [INFO] - Binding failed, this device has been forced to verify the account qualification by Xiaomi. (30001)
```

:(

changyun233, 2023-11-25:
> Setting `$id` to empty string worked for me. However, it says
> 
> ```
> [2023-11-24] [15:06:50] [INFO] - Binding failed, this device has been forced to verify the account qualification by Xiaomi. (30001)
> ```
> 
> :(

same 30001 for me.

Device :Xiaomi 14 with 1.0.23.11.21.DEV

I had tried to apply for BL unlock in Xiaomishequ. Obviously I did not pass...

Jesus is there a Chinese brand Android phone with loose unlock policy...

closed 2023-11-26 by MlgmXyysd