MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #12: 能否出个视频教程

aakk007, 2023-12-08:
现有的文字描述对不是太了解相关技术的人来说无法操作，能否出个视频教程？

aakk007, 2023-12-08:
尝试成功了

171281898, 2023-12-17:
> 现有的文字描述对不是太了解相关技术的人来说无法操作，能否出个视频教程？

> 尝试成功了

可以教教我吗？

Aaminly, 2023-12-31:
一直提示这个   invalid transport id
[image]

A1ja, 2024-01-02:
> 尝试成功了

您好，可以帮我弄我一下嘛，手机是红米k70，不白嫖，方便的话给我一个联系方式

bwn007, 2024-01-05:
[2024-01-05] [06:52:15] [INFO] - Waiting for device connection...一直提示这个 什么原因

zzzztao, 2024-01-05:
> [2024-01-05] [06:52:15] [INFO] - Waiting for device connection...一直提示这个 什么原因

电脑没有识别到手机

dex2oat, 2024-01-11:
> 一直提示这个 invalid transport id <img alt="image" width="920" src="https://private-user-images.githubusercontent.com/28622560/293512237-aa1fd5b7-559a-41ed-9eb5-347bcc368ea9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NTczNzksIm5iZiI6MTcwNDk1NzA3OSwicGF0aCI6Ii8yODYyMjU2MC8yOTM1MTIyMzctYWExZmQ1YjctNTU5YS00MWVkLTllYjUtMzQ3YmNjMzY4ZWE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTExVDA3MTExOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhOTE2M2E1YmNhMDU0MTU2MGFkNDRiNTljNTNiNzBlMjUyMjIwODNmMGI5ODlhMTQwNjAyYjJlY2FiZjk4ZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zlYKEd4YC5Fe1InNbtXQV6c4PR1HmpjIXCrSmsPdKbk">

代码里改一下吧，bypass.php第279行，true改false，意思不用adb的-t用-s,另一个adb.php里面有写。

LeoWang40, 2024-02-22:
想問樓主，是否有視頻教學影片可參考，感激不盡

3193016372, 2024-05-20:
大陆的14pro可以么

LeoWang40, 2024-06-13:
可以