MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #13: cannot making POST request

RuisongZhou, 2023-12-11:
the log is 
```
(base) ☁  Bypass-1.0-fix-universal  php bypass.php
[2023-12-11] [03:16:12] [INFO] - ************************************
[2023-12-11] [03:16:12] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2023-12-11] [03:16:12] [INFO] - * By NekoYuzu          Version 1.0 *
[2023-12-11] [03:16:12] [INFO] - ************************************
[2023-12-11] [03:16:12] [INFO] - GitHub: https://github.com/MlgmXyysd
[2023-12-11] [03:16:12] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2023-12-11] [03:16:12] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2023-12-11] [03:16:12] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2023-12-11] [03:16:12] [INFO] - My Blog: https://www.neko.ink/
[2023-12-11] [03:16:12] [INFO] - ************************************
[2023-12-11] [03:16:12] [INFO] - Starting ADB server...
[2023-12-11] [03:16:12] [INFO] - Processing device 1affaa10(device:houji)...
[2023-12-11] [03:16:12] [INFO] - Processing device id ...
[2023-12-11] [03:16:12] [INFO] - Finding BootLoader unlock bind request...
[2023-12-11] [03:16:12] [INFO] * Now you can bind account in the developer options.
[2023-12-11] [03:16:12] [INFO] - "/Users/ruisongzhou/Downloads/Bypass-1.0-fix-universal/libraries/adb-darwin" logcat *:S CloudDeviceStatus:V
[2023-12-11] [03:16:20] [INFO] - Account bind request found! Let's block it.
[2023-12-11] [03:16:20] [INFO] - Refactoring parameters...
[2023-12-11] [03:16:20] [INFO] - Sending POST request...
[2023-12-11] [03:16:23] [INFO] ! Fail to send request, check your internet connection.
```
My XiaoMi14 and my PC is connect to Internet, and I also tried to connect to the internet with or without using VPN, but failed. The phone is not open WIFI and only use cellular network

MlgmXyysd, 2023-12-11:
retry

RuisongZhou, 2023-12-12:
I have sovled this problem. The key point is that when use USB connect to your PC, it cannot be set to **USB network sharing mode**, otherwise the POST request can not be successd.

closed 2023-12-12 by RuisongZhou