MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #15: 提示错误[2023-12-14] [00:10:26] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in D:\php\php-8.3.0-Win32-vs16-x64\bypass.php:230
Stack trace:
#0 D:\php\php-8.3.0-Win32-vs16-x64\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in D:\php\php-8.3.0-Win32-vs16-x64\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in D:\php\php-8.3.0-Win32-vs16-x64\bypass.php:230
Stack trace:
#0 D:\php\php-8.3.0-Win32-vs16-x64\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in D:\php\php-8.3.0-Win32-vs16-x64\bypass.php on line 230

Shadow071107, 2023-12-14:


VictorQR, 2023-12-14:
我也是提示报错

VictorQR, 2023-12-14:
> 我也是提示报错PHP Warning: PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (鎵句笉鍒版寚瀹氱殑妯″潡銆?, C:\php\ext\php_curl.dll (鎵句笉鍒版寚瀹氱殑妯″ 潡銆?) in Unknown on line 0
> 
> Warning: PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (鎵句笉鍒版寚瀹氱殑妯″潡銆?, C:\php\ext\php_curl.dll (鎵句笉鍒版寚瀹氱殑妯″潡銆?) in Unknown on line 0 PHP Warning: PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (鎵句笉鍒版寚瀹氱殑妯″潡 銆?, C:\php\ext\php_openssl.dll (鎵句笉鍒版寚瀹氱殑妯″潡銆?) in Unknown on line 0
> 
> Warning: PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (鎵句笉鍒版寚瀹氱殑妯″潡銆?, C:\php\ext\php_openssl.dll (鎵句笉鍒版寚瀹氱殑妯″潡銆?) in Unknown on line 0 [2023-12-14] [02:50:41] [INFO] - ************************************ [2023-12-14] [02:50:41] [INFO] - * Xiaomi HyperOS BootLoader Bypass * [2023-12-14] [02:50:41] [INFO] - * By NekoYuzu Version 1.0 * [2023-12-14] [02:50:41] [INFO] - ************************************ [2023-12-14] [02:50:41] [INFO] - GitHub: https://github.com/MlgmXyysd [2023-12-14] [02:50:41] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637 [2023-12-14] [02:50:41] [INFO] - X (Twitter): https://x.com/realMlgmXyysd [2023-12-14] [02:50:41] [INFO] - PayPal: https://paypal.me/MlgmXyysd [2023-12-14] [02:50:41] [INFO] - My Blog: https://www.neko.ink/ [2023-12-14] [02:50:41] [INFO] - ************************************ [2023-12-14] [02:50:41] [INFO] - Starting ADB server... [2023-12-14] [02:50:41] [INFO] - Waiting for device connection... [2023-12-14] [02:50:42] [INFO] - Waiting for device connection... [2023-12-14] [02:50:43] [INFO] - Waiting for device connection... [2023-12-14] [02:50:44] [INFO] - Waiting for device connection... [2023-12-14] [02:50:45] [INFO] - Waiting for device connection... [2023-12-14] [02:50:47] [INFO] - Processing device 488fb5a3(4)... [2023-12-14] [02:50:47] [INFO] - Finding BootLoader unlock bind request... [2023-12-14] [02:50:47] [INFO] * Now you can bind account in the developer options. [2023-12-14] [02:50:53] [INFO] - Account bind request found! Let's block it. [2023-12-14] [02:50:53] [INFO] - Refactoring parameters... PHP Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in D:\Desktop\php-8.3.0-nts-Win32-vs16-x64\bypass.php:230 Stack trace: #0 D:\Desktop\php-8.3.0-nts-Win32-vs16-x64\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...') #1 {main} thrown in D:\Desktop\php-8.3.0-nts-Win32-vs16-x64\bypass.php on line 230
> 
> Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in D:\Desktop\php-8.3.0-nts-Win32-vs16-x64\bypass.php:230 Stack trace: #0 D:\Desktop\php-8.3.0-nts-Win32-vs16-x64\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...') #1 {main} thrown in D:\Desktop\php-8.3.0-nts-Win32-vs16-x64\bypass.php on line 230 请问如何解决的？

没事了，redmi k70这种出厂澎湃OS设备不行

Shadow071107, 2023-12-14:
> 没事了，redmi k70这种出厂澎湃OS设备不行

我是k60。。。

Remember9, 2023-12-14:
windows系统，修改php.ini文件，在文件里搜索extension_dir定位到 extension_dir = "ext"这行，把这行前边用来注释的分号删除

closed 2023-12-19 by MlgmXyysd