MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #20: Need help. I'm trying to open the bootloader for Xiaomi pad 6.

wewdas2, 2023-12-25:
Need help opening the bootloader for the tablet.  I'm trying to open the bootloader for Xiaomi pad 6 max with HyperOS([1.0.2.0](https://1.0.2.0/).UMHCNXM) version.  Perhaps I didn't take something into account in the instructions.  I need a hint.  I would be very grateful.
[image]
[image]
[image]

SanGongHappy, 2023-12-25:
You can try copying the 'php.ini-development' file and renaming it to 'php.ini'. Although I ultimately failed, you seem to encounter a problem similar to yours.

MlgmXyysd, 2023-12-25:
> Download [platform-tools](https://developer.android.com/studio/releases/platform-tools) and place them in **libraries**.

TechNo1geek, 2023-12-30:
I get the exact same results

TechNo1geek, 2023-12-30:
`PS C:\Users\techno1geek\Desktop\xiaomi14\Xiaomi-HyperOS-BootLoader-Bypass-master\Xiaomi-HyperOS-BootLoader-Bypass-master\libraries> php adb.php
PHP Warning:  PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (The specified module could not be found), C:\php\ext\php_curl.dll (The specified module could not be found)) in Unknown on line 0

Warning: PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (The specified module could not be found), C:\php\ext\php_curl.dll (The specified module could not be found)) in Unknown on line 0
PHP Warning:  PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (The specified module could not be found), C:\php\ext\php_openssl.dll (The specified module could not be found)) in Unknown on line 0

Warning: PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (The specified module could not be found), C:\php\ext\php_openssl.dll (The specified module could not be found)) in Unknown on line 0
PS C:\Users\techno1geek\Desktop\xiaomi14\Xiaomi-HyperOS-BootLoader-Bypass-master\Xiaomi-HyperOS-BootLoader-Bypass-master\libraries>`

deusmosk81, 2023-12-31:
> `PS C:\Users\techno1geek\Desktop\xiaomi14\Xiaomi-HyperOS-BootLoader-Bypass-master\Xiaomi-HyperOS-BootLoader-Bypass-master\libraries> php adb.php Предупреждение PHP: при запуске PHP: не удается загрузить динамическую библиотеку 'curl' (пробовал: C:\php\ext\curl (Указанный модуль не найден), C:\php\ext\php_curl.dll (Указанный модуль не найден)) в Unknown в строке 0
> 
> Предупреждение: При запуске PHP: не удается загрузить динамическую библиотеку 'curl' (пробовал: C:\php\ext\curl (Указанный модуль не найден), C:\php\ext\php_curl.dll (Указанный модуль не найден)) в Unknown в строке 0 Предупреждение PHP: Запуск PHP: не удается загрузить динамическую библиотеку 'openssl' (пробовал: C:\php\ext\openssl (Указанный модуль не найден), C:\php\ext\php_openssl.dll (Указанный модуль не найден)) в Unknown в строке 0
> 
> Предупреждение: При запуске PHP: не удается загрузить динамическую библиотеку 'openssl' (пробовал: C:\php\ext\openssl (Указанный модуль не найден), C:\php\ext\php_openssl.dll (Указанный модуль не найден)) в Unknown в строке 0 PS C:\Users\techno1geek\Desktop\xiaomi14\Xiaomi-HyperOS-BootLoader-Bypass-master\Xiaomi-HyperOS-BootLoader-Bypass-master\libraries>`

To solve the problem in Windows, you need to specify the path to the libraries in php.ini

[image]

[image]

wewdas2, 2024-01-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
I'm asking again. I made a libraries folder and pushed the platform-tools there. Maybe a problem with the adb driver?

MlgmXyysd, 2024-01-19:
> I'm trying again. I made a libraries folder and put the platform tools there. Maybe there is a problem with the adb driver?

https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/29#issuecomment-1900298533

wewdas2, 2024-01-20:
https://youtu.be/H674oTdztWU?si=i8PFHu4u92bzewl5
Hmmm...