MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #22: 按操作一直报：Waiting for device connection...

opser-gavin, 2023-12-31:
如标题，我完全按照说明操作的。
到执行脚本那一步，一直报：Waiting for device connection...
用的小米原装的数据线，开发者选项开户OEM解锁，USB调试全开启了已经
[image]

opser-gavin, 2023-12-31:
选中始终允许来自此计算机的调试，然后单击确定。
这一步我连接手机，没有提示这个。难道是因为这个吗？
盼回~~

opser-gavin, 2023-12-31:
不管怎么操作 都是一直刷 这个。。。。。。。。。。。。。。。。。。。。。。
[2023-12-31] [03:26:25] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:26] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:27] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:28] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:29] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:30] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:31] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:32] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:33] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:34] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:35] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:37] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:38] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:39] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:40] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:41] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:42] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:43] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:44] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:45] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:46] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:47] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:48] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:49] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:50] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:51] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:52] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:53] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:54] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:55] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:56] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:57] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:58] [INFO] - Waiting for device connection...
[2023-12-31] [03:26:59] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:00] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:01] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:02] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:03] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:04] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:05] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:06] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:07] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:08] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:09] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:10] [INFO] - Waiting for device connection...
[2023-12-31] [03:27:11] [INFO] - Waiting for device connection...

opser-gavin, 2023-12-31:
[image]

请问这算不是算成功了呀？ 但是我在开发者选项中看解锁状态还和以前一样：“绑定失败，请前往小米社区中申请授权后重试“

opser-gavin, 2023-12-31:
[image]
好像可以了，感谢大佬~~

bwn007, 2024-01-05:
> 如标题，我完全按照说明操作的。 到执行脚本那一步，一直报：Waiting for device connection... 用的小米原装的数据线，开发者选项开户OEM解锁，USB调试全开启了已经 [...]

请问你是如何解决的

Itsusinn, 2024-01-05:
> 请问你是如何解决的

换usb线，原装线疑似有问题

MlgmXyysd, 2024-01-19:
> 选中始终允许来自此计算机的调试，然后单击确定。 这一步我连接手机，没有提示这个。难道是因为这个吗？ 盼回~~

说明设备没连接成功，检查驱动或者电脑接口或者数据线或者手机尾插，非脚本问题

closed 2024-01-19 by MlgmXyysd