MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #30: adb-darwin: invalid transport id

ekanshsinghal, 2024-01-17:
`[2024-01-17] [10:59:37] [INFO] - ************************************
[2024-01-17] [10:59:37] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-01-17] [10:59:37] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-01-17] [10:59:37] [INFO] - ************************************
[2024-01-17] [10:59:37] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-01-17] [10:59:37] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-01-17] [10:59:37] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-01-17] [10:59:37] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-01-17] [10:59:37] [INFO] - My Blog: https://www.neko.ink/
[2024-01-17] [10:59:37] [INFO] - ************************************
[2024-01-17] [10:59:37] [INFO] - Starting ADB server...
[2024-01-17] [10:59:38] [INFO] - Processing device 85986346(device:liuqin)...
[2024-01-17] [10:59:38] [INFO] - Finding BootLoader unlock bind request...
[2024-01-17] [10:59:38] [INFO] * Now you can bind account in the developer options.
adb-darwin: invalid transport id
[2024-01-17] [10:59:38] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /Users/ekansh/bypass.php on line 330 and defined in /Users/ekansh/bypass.php:228
Stack trace:
#0 /Users/ekansh/bypass.php(330): decryptData(NULL)
#1 {main}
  thrown in /Users/ekansh/bypass.php on line 228

Fatal error: Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /Users/ekansh/bypass.php on line 330 and defined in /Users/ekansh/bypass.php:228
Stack trace:
#0 /Users/ekansh/bypass.php(330): decryptData(NULL)
#1 {main}
  thrown in /Users/ekansh/bypass.php on line 228
`

dex2oat, 2024-01-18:
> ```
> [2024-01-17] [10:59:37] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
> [2024-01-17] [10:59:37] [INFO] - * By NekoYuzu          Version 1.0 *
> [2024-01-17] [10:59:37] [INFO] - [...]

代码里改一下吧，bypass.php第279行，true改false，意思是不用adb的-t用-s，另一个adb.php里面有写。

dex2oat, 2024-01-19:
> ```
> unlock tool.
> ```

unlock tool,"https://en.miui.com/unlock/index.html",It shows waiting for 168 hours, which means it is successful. It can be unlocked after 168 hours

dex2oat, 2024-01-19:
> wait time can be viewed in the unlock tool.

unlock tool,"[https://en.miui.com/unlock/index.html",It](https://en.miui.com/unlock/index.html%22,It) shows waiting for 168 hours, which means it is successful. It can be unlocked after 168 hours

MlgmXyysd, 2024-01-19:
It is normal to show binding failure in the device, please use the unlocking tool to check

tux-86, 2024-02-13:
Same bug is also in linux, but with a code change suggested by dex2oat, I finally got the script working.