MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #31: Fail

MercyTHegoat, 2024-01-18:
When i try to bind account,script is saying me "fail. check internet connection" what to do?

MlgmXyysd, 2024-01-18:
check your network connection, or try to change server region to global

MercyTHegoat, 2024-01-18:
not working(

MlgmXyysd, 2024-01-19:
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/29#issuecomment-1899935290

MercyTHegoat, 2024-01-19:
i dont know,he still saying about internet. i will check what with my internet. there is waht he say:

[2024-01-19] [19:32:11] [INFO] - ************************************
[2024-01-19] [19:32:11] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-01-19] [19:32:11] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-01-19] [19:32:11] [INFO] - ************************************
[2024-01-19] [19:32:11] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-01-19] [19:32:11] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-01-19] [19:32:11] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-01-19] [19:32:11] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-01-19] [19:32:11] [INFO] - My Blog: https://www.neko.ink/
[2024-01-19] [19:32:11] [INFO] - ************************************
[2024-01-19] [19:32:11] [INFO] - Starting ADB server...
[2024-01-19] [19:32:12] [INFO] - Processing device 49aalb5dkzifswqw(1)...
[2024-01-19] [19:32:12] [INFO] - Finding BootLoader unlock bind request...
[2024-01-19] [19:32:12] [INFO] * Now you can bind account in the developer options.
[2024-01-19] [19:32:23] [INFO] - Account bind request found! Let's block it.
[2024-01-19] [19:32:23] [INFO] - Refactoring parameters...
string(1004) "8/8Mnp7WcRph8sbnR99N7oUMFTAhPgS8Nt5hrQpKsVyitMrnMNWIZ1LmWOIZKAHPnEDWLgEAv4xdwNdNS+39hHwdKCgFHAnseM/Js0UmaTJ4iZeE9nuP/MJg3i+p3I3tM3f3+tm5o0sxhxWFFzyUOz78WNMlZOcj2lIcU6PC+Zl80EjW8FiZE3MhqwdY6jOqQLkXQylWi3yhobug2DNJrTRqfTKGd7mfLbKFGZCnGZWt0Khc1wP0ZtkIPZeJRe1TskKHirBCDfr+N0SdUUSW549D5So39O7TRfctsCvD6PnQyqF8lkT1X5VRYTJt0M/6hz1I7R/Cn8aKuAt3BggzE6cggkhNJi7H9gERoBWZ4D7IwCRJKNxPk9cMn9/xkhi7Xjm0hKGG3dsvUdvTv9dELTKxeFOzB0ehEQvACdxN8Aa+CVVwMoq3H+Q/wVQK8DlhAQytdJleIXds00UM7/4h8QOjOUVgvxR8RzT4Vu25AzAdTuX70PHE1TnG24B8RnJ0v/ZkSVg5gzBmH+WPU+Ii+TC2Rlq25+RfttgLTYS2ClRXKYTVfhzZiy5h44jpTv1TsMuemMZZM4fG1CdF1/qMgKISHwpbE/5lFiebFEgtFwXeh8Wbs5NwkMLgj01bj4OxrVceD7F6n+gYE78A1izH2f2BYu3KbbQD0QyEmEWUl7QiJa6QUDNlNw55bk32lRcJK//bKKnKFSkL45IXbqD9VI3ZyWpYgDJNldQ7Xd/WSw+E2orI2DbCczSnzFHKUwJTT8wILh04vhWV4AzE6Ev7EDy2MzwYRoavZRJX1K4xHMSyXfpxkm1miRMk8IRQf0IIx1xXPV1D7IBSBjIC2lv8sa29TRkxvX0r5qYdlNz8RUjzD5DPGejoOryQl30AhGXOOZFNozKN9RnPO+8vNOLL+0UAbSf31UPBzrHKJCVKvDf6KmEH0r80FY5N+711p2gcLm3LBmRYcj++dail1qKGcwY+vz4zLCxYHy3T9wo0MiY="
[2024-01-19] [19:32:23] [INFO] - Sending POST request...
[2024-01-19] [19:32:23] [INFO] ! Fail to send request, check your internet connection.

MercyTHegoat, 2024-01-26:
> данные могут быть расшифрованы, поэтому проблема вызвана не этим. попробуйте зайти на адрес сервера разблокировки через браузер или с помощью 

> данные могут быть расшифрованы, поэтому проблема вызвана не этим. попробуйте зайти на адрес сервера разблокировки через браузер или с помощью ping

thank you so much,it works

closed 2024-01-26 by MercyTHegoat