MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #35: Request parameter error (10000)

LiuXuan-bk, 2024-01-22:
设备型号 Redmi K60 Pro
系统版本： 1.0.5.0.UMKCNXM
提示Now you can bind account in the developer options.后点击绑定账号和设备，控制台输出 Request parameter error (10000)
查看历史issue后尝试降级系统，尝试卡刷和线刷都失败了
以下是控制台内容
[2024-01-22] [13:09:47] [INFO] - ************************************
[2024-01-22] [13:09:47] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-01-22] [13:09:47] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-01-22] [13:09:47] [INFO] - ************************************
[2024-01-22] [13:09:47] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-01-22] [13:09:47] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-01-22] [13:09:47] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-01-22] [13:09:47] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-01-22] [13:09:47] [INFO] - My Blog: https://www.neko.ink/
[2024-01-22] [13:09:47] [INFO] - ************************************
[2024-01-22] [13:09:47] [INFO] - Starting ADB server...
[2024-01-22] [13:09:49] [INFO] - Processing device fb44ea77(8)...
[2024-01-22] [13:09:49] [INFO] - Finding BootLoader unlock bind request...
[2024-01-22] [13:09:50] [INFO] * Now you can bind account in the developer options.
[2024-01-22] [13:09:59] [INFO] - Account bind request found! Let's block it.
[2024-01-22] [13:09:59] [INFO] - Refactoring parameters...
[2024-01-22] [13:09:59] [INFO] - Sending POST request...
[2024-01-22] [13:10:00] [INFO] - Request parameter error (10000)
请问有什么解决方案吗？

MlgmXyysd, 2024-01-22:
Duplicate #29
https://twitter.com/realMlgmXyysd/status/1748386904050110675

LiuXuan-bk, 2024-01-22:
抱歉，再次阅读发现，需要覆盖安装setting，覆盖安装后重试，问题已经解决

closed 2024-01-22 by LiuXuan-bk

sxakhyct, 2024-03-21:
这个成功解决了么？

sxakhyct, 2024-03-21:
> > 抱歉，再次阅读发现，需要覆盖安装setting，覆盖安装后重试，问题已经解决
> 
> 怎么弄？

我从miui12.5里面提取了个设置，在我的澎湃OS里安装的时候，提示不允许降级安装。请问你怎么安装的啊？

woody0116, 2024-03-25:
Hi MlgmXyysd ,
    I use [MlgmXyysd](https://github.com/MlgmXyysd) solution but still fail , my case is "redmi note 13 5G " , I didn't find any source ( 1000 error) for  "redmi note 13 5G " ， it's  fail to try to use "小米13_13Pro_13Ultra"  source . Has the case solution to be fixed now ?