MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #37: 红米note13 pro 返回The system is being upgraded. Please try again later. (30003)

cy123, 2024-01-23:
[image]

MlgmXyysd, 2024-01-23:
字面意思，服务器维护

closed 2024-01-25 by MlgmXyysd

ilharp, 2024-01-29:
目测小米为保证 HyperOS 用户量已经停掉了 Note 13 全系的 BL 权限，咨询小米商城客服说没有恢复时间。小米社区从去年 11 月起每天都有人反馈无法解锁

我也没办法，最后今天找淘宝大神解了

cy123, 2024-01-29:
> 目测小米为保证 HyperOS 用户量已经停掉了 Note 13 全系的 BL 权限，咨询小米商城客服说没有恢复时间。小米社区从去年 11 月起每天都有人反馈无法解锁
> 
> 我也没办法，最后今天找淘宝大神解了

是刷机降级到miui13吗？花了多少？

ilharp, 2024-01-29:
> > 目测小米为保证 HyperOS 用户量已经停掉了 Note 13 全系的 BL 权限，咨询小米商城客服说没有恢复时间。小米社区从去年 11 月起每天都有人反馈无法解锁
> > 我也没办法，最后今天找淘宝大神解了
> 
> 是刷机降级到miui13吗？花了多少？

出厂 MIUI 14，没有升降级。花了 285

cy123, 2024-01-31:
> > > 目测小米为保证 HyperOS 用户量已经停掉了 Note 13 全系的 BL 权限，咨询小米商城客服说没有恢复时间。小米社区从去年 11 月起每天都有人反馈无法解锁
> > > 我也没办法，最后今天找淘宝大神解了
> > 
> > 
> > 是刷机降级到miui13吗？花了多少？
> 
> 出厂 MIUI 14，没有升降级。花了 285

我也花钱解了，我还想看他怎么解了，一不注意就完成了，不知道是内部人员留的后门，还是利用了什么漏洞

SergChe88, 2024-03-07:
> 字面意思，服务器维护，你在手机绑定也一样提示

Вы сопровождаете свой git таким описанием: "A PoC that exploits a vulnerability to bypass the Xiaomi HyperOS community restrictions of BootLoader unlocked account bindings."

При этом, вы пишете, что такое же сообщение появляется при использовании стандартного метода привязки. Так чем ваш метод отличается от стандартного? Где вы применяете "exploits a vulnerability"?

16778738, 2025-04-27:
各位大佬,现在还有什么路径可以解锁bl嘛?