MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #40: Fail to send request, check your internet connection.

NVG-064, 2024-01-25:
Stuck on "Fail to send request, check your internet connection".

Then I found that $data was null. Value of $args or $headers exists, but it can't decrypt the $args or $headers. Here's the result.

==================================================================================

[2024-01-25] [10:53:08] [INFO] - Finding BootLoader unlock bind request...
[2024-01-25] [10:53:08] [INFO] * Now you can bind account in the developer options.
[2024-01-25] [10:53:14] [INFO] - Output: --------- beginning of main

[2024-01-25] [10:53:14] [INFO] - Output: 01-25 17:51:12.953  3676 12357 D CloudDeviceStatus: stateCode: 200

[2024-01-25] [10:53:14] [INFO] - Output: 01-25 17:51:12.953  3676 12357 D CloudDeviceStatus: content: {"result":"ok","retriable":false,"code":0,"data":{"nonce":"AgUAG0AnL1sToDRfZjQ2YzJmZDSPx0BS5HAXQlmBVpbnd1-nmbZqqQ"},"description":"成功","ts":1706176272866}

[2024-01-25] [10:53:15] [INFO] - Output: 01-25 17:51:13.166  3676 12357 I CloudDeviceStatus: args: #&^u2V88RBjxOXSU7roD/JBFDtMI9690u6p2zYKruQLbWDdw76Am9man7Y9WRPDgCRQx1nBj4ugxH4c6xpTd1lX7ZTaLAZab/qCHFNp3PZr8x75zUDF0KyFDGr689GSMqIYc8meBrewK7aj8T+eTsfnBFMyZj8qck+IBBU0+LyQHlW4iQAkBKg6qrLyP5rnLGQ51QxM4I0P+x6wlwREk9MoSukl2pRyS6AwL+f4S6riySxxkH0lENQ/Dwsg4hjODo6XAhTYDW5z5iyLJrGlgDOon73ZCQrAnAdxfpIndj1g3+SRtVmG0PNxMA1y0itGfnltgbDeXTXhyRmr+GsQNe/vNQ==!!Dhoeo7t00/HT9DrsOU0u6IjrmXnlNvfnoYGIxESx1is1MfyS5D/qjE6m/12ARdZF4+Gy5s/gV0gwtF0e+0PgotMv7eH18Ozb+bQpoeNSepM6/PX4YiOovFa7+RujJmSzqlsCOkBTZd28WMCS1128VRL/UQFQ4LXnKHg7GmQtZLxr8rQQQuNgecUzGlkK1U/qqAssBCbmLX5cgemUrfwDFDfqwN5LiuYmgBGymfg2YFyeNYcSmsOem848WaW5FeWvoyKi24pLc098jhwE8NQvUz0XZay5Hk9jCln2dGmLsbYjNHsS5Z8NDN/Zj5k6uzT9oqWdmaj/Fd0l5+8TYnQ211D9JZ4Qx8qV1BKZjhVRx/CVvI1C784gpLO5SccAMtcmY3YM3DoWQPbLbQFcEEEen7545MDakM1NdPEkO4iUOKLBfMnKgIStW7IKmtNUB1INtmwbung7aE5aVRohEskatHLVAtJ0Rt+Lfiv4eF5zxo5G1m47KKa2rM4ovYA8QQPCCULThkbIDA168StpzKa8p6YeqjG1n/m9ie2UNBYJTQZ96AFm+VXX2VyeeiUSM2FPWwSo8BtBBNYKiFe0AO3HXx/HxFW/6YZWO96c65NWx8cYCfzaL3oBkY+wI3dujx1/jC0DaqZ/ALRLAad9sKi6Mk+nLuKlgVWDEc84CIe2KuC36qIJyHI+a+5XUm5Kg5H0zE0nXV4O59qXiL9xwpphtjMeo9A0cbHMo5yy14ajmqLnRVPhVaUMOz6Z2fhU/YSRI7076fanze8wxmPTSIqFr7r/t3M01D6u02+myTT2o5HX6jPsDsqhgTJeBTvS+yFWUk8ETEFx0o/DrzR7GnSgQd/8872h+oiu98iRbzpOBYlO2v95JMLWZIVPmlL6sd0MSm7+RwdBM7rHMTIy31YTpg==^&#

[2024-01-25] [10:53:15] [INFO] - Args: #&^u2V88RBjxOXSU7roD/JBFDtMI9690u6p2zYKruQLbWDdw76Am9man7Y9WRPDgCRQx1nBj4ugxH4c6xpTd1lX7ZTaLAZab/qCHFNp3PZr8x75zUDF0KyFDGr689GSMqIYc8meBrewK7aj8T+eTsfnBFMyZj8qck+IBBU0+LyQHlW4iQAkBKg6qrLyP5rnLGQ51QxM4I0P+x6wlwREk9MoSukl2pRyS6AwL+f4S6riySxxkH0lENQ/Dwsg4hjODo6XAhTYDW5z5iyLJrGlgDOon73ZCQrAnAdxfpIndj1g3+SRtVmG0PNxMA1y0itGfnltgbDeXTXhyRmr+GsQNe/vNQ==!!Dhoeo7t00/HT9DrsOU0u6IjrmXnlNvfnoYGIxESx1is1MfyS5D/qjE6m/12ARdZF4+Gy5s/gV0gwtF0e+0PgotMv7eH18Ozb+bQpoeNSepM6/PX4YiOovFa7+RujJmSzqlsCOkBTZd28WMCS1128VRL/UQFQ4LXnKHg7GmQtZLxr8rQQQuNgecUzGlkK1U/qqAssBCbmLX5cgemUrfwDFDfqwN5LiuYmgBGymfg2YFyeNYcSmsOem848WaW5FeWvoyKi24pLc098jhwE8NQvUz0XZay5Hk9jCln2dGmLsbYjNHsS5Z8NDN/Zj5k6uzT9oqWdmaj/Fd0l5+8TYnQ211D9JZ4Qx8qV1BKZjhVRx/CVvI1C784gpLO5SccAMtcmY3YM3DoWQPbLbQFcEEEen7545MDakM1NdPEkO4iUOKLBfMnKgIStW7IKmtNUB1INtmwbung7aE5aVRohEskatHLVAtJ0Rt+Lfiv4eF5zxo5G1m47KKa2rM4ovYA8QQPCCULThkbIDA168StpzKa8p6YeqjG1n/m9ie2UNBYJTQZ96AFm+VXX2VyeeiUSM2FPWwSo8BtBBNYKiFe0AO3HXx/HxFW/6YZWO96c65NWx8cYCfzaL3oBkY+wI3dujx1/jC0DaqZ/ALRLAad9sKi6Mk+nLuKlgVWDEc84CIe2KuC36qIJyHI+a+5XUm5Kg5H0zE0nXV4O59qXiL9xwpphtjMeo9A0cbHMo5yy14ajmqLnRVPhVaUMOz6Z2fhU/YSRI7076fanze8wxmPTSIqFr7r/t3M01D6u02+myTT2o5HX6jPsDsqhgTJeBTvS+yFWUk8ETEFx0o/DrzR7GnSgQd/8872h+oiu98iRbzpOBYlO2v95JMLWZIVPmlL6sd0MSm7+RwdBM7rHMTIy31YTpg==^&#
PHP Warning:  Array to string conversion in D:\php\Bypass\bypass.php on line 319
[2024-01-25] [10:53:15] [INFO] - Matches: Array
[2024-01-25] [10:53:15] [INFO] - Output: 01-25 17:51:13.167  3676 12357 I CloudDeviceStatus: headers: #&^u2V88RBjxOXSU7roD/JBFDtMI9690u6p2zYKruQLbWDdw76Am9man7Y9WRPDgCRQx1nBj4ugxH4c6xpTd1lX7ZTaLAZab/qCHFNp3PZr8x75zUDF0KyFDGr689GSMqIYc8meBrewK7aj8T+eTsfnBFMyZj8qck+IBBU0+LyQHlW4iQAkBKg6qrLyP5rnLGQ51QxM4I0P+x6wlwREk9MoSukl2pRyS6AwL+f4S6riySxxkH0lENQ/Dwsg4hjODo6XAhTYDW5z5iyLJrGlgDOon73ZCQrAnAdxfpIndj1g3+SRtVmG0PNxMA1y0itGfnltgbDeXTXhyRmr+GsQNe/vNQ==!!5u/PW84zNpUNPOxuXg5r9I0klPPOCK1GkwiR6DYsYsWhEQlLBsvLaLxD9w7MIubZHLHreF1qgYmZ27OQW4oLSVE73iZdeEJTUIAnxv1SRJ69PstSfE+emBrjKoh4AaRzRvbGIXNn0gzXNJUOG+vydKkN2XPmG0xyeRVy0RC6Pbj98G0v0CE7aNUv3sLcvY6qxScVpxgArku5yHRnwPzn+BPpcaZHmkcQZqDQGPs3Fv2OkufxV4112YDuHOZZAAtPzEDN1W2PLcaP0dP2bWMWWl4LHu/S4Gi50ST+uMyv+InPurXEWcbYMpTedIrYfSemV1xjRQPWj1fcZ6EMMlhAFs2t9r2U/QVkT/K4Xm7IBmilC//CJ0MyJCWxyP7NHLhhsMLxN5idEpzWiN9X87uldthA/CqkW+bNkYa3lMqHQbMWVI8Woh5nR4pfWmkFpmEcpAQ48ZPpvRLLjqtMfNLWbBRE0ya8iufG21qAv28KErzKoOPWXFCgl5qynAlps57M/1G0N4C+r+u6CFb8ULpLElcpoVhZyDXEL8LzCRnYVhc=^&#

[2024-01-25] [10:53:15] [INFO] - Headers: #&^u2V88RBjxOXSU7roD/JBFDtMI9690u6p2zYKruQLbWDdw76Am9man7Y9WRPDgCRQx1nBj4ugxH4c6xpTd1lX7ZTaLAZab/qCHFNp3PZr8x75zUDF0KyFDGr689GSMqIYc8meBrewK7aj8T+eTsfnBFMyZj8qck+IBBU0+LyQHlW4iQAkBKg6qrLyP5rnLGQ51QxM4I0P+x6wlwREk9MoSukl2pRyS6AwL+f4S6riySxxkH0lENQ/Dwsg4hjODo6XAhTYDW5z5iyLJrGlgDOon73ZCQrAnAdxfpIndj1g3+SRtVmG0PNxMA1y0itGfnltgbDeXTXhyRmr+GsQNe/vNQ==!!5u/PW84zNpUNPOxuXg5r9I0klPPOCK1GkwiR6DYsYsWhEQlLBsvLaLxD9w7MIubZHLHreF1qgYmZ27OQW4oLSVE73iZdeEJTUIAnxv1SRJ69PstSfE+emBrjKoh4AaRzRvbGIXNn0gzXNJUOG+vydKkN2XPmG0xyeRVy0RC6Pbj98G0v0CE7aNUv3sLcvY6qxScVpxgArku5yHRnwPzn+BPpcaZHmkcQZqDQGPs3Fv2OkufxV4112YDuHOZZAAtPzEDN1W2PLcaP0dP2bWMWWl4LHu/S4Gi50ST+uMyv+InPurXEWcbYMpTedIrYfSemV1xjRQPWj1fcZ6EMMlhAFs2t9r2U/QVkT/K4Xm7IBmilC//CJ0MyJCWxyP7NHLhhsMLxN5idEpzWiN9X87uldthA/CqkW+bNkYa3lMqHQbMWVI8Woh5nR4pfWmkFpmEcpAQ48ZPpvRLLjqtMfNLWbBRE0ya8iufG21qAv28KErzKoOPWXFCgl5qynAlps57M/1G0N4C+r+u6CFb8ULpLElcpoVhZyDXEL8LzCRnYVhc=^&#
PHP Warning:  Array to string conversion in D:\php\Bypass\bypass.php on line 330
[2024-01-25] [10:53:15] [INFO] - Matches: Array
[2024-01-25] [10:53:15] [INFO] - Account bind request found! Let's block it.
[2024-01-25] [10:53:15] [INFO] - Refactoring parameters...
[2024-01-25] [10:53:15] [INFO] - Init:
[2024-01-25] [10:53:15] [INFO] - Data:
[2024-01-25] [10:53:15] [INFO] - Args: #&^u2V88RBjxOXSU7roD/JBFDtMI9690u6p2zYKruQLbWDdw76Am9man7Y9WRPDgCRQx1nBj4ugxH4c6xpTd1lX7ZTaLAZab/qCHFNp3PZr8x75zUDF0KyFDGr689GSMqIYc8meBrewK7aj8T+eTsfnBFMyZj8qck+IBBU0+LyQHlW4iQAkBKg6qrLyP5rnLGQ51QxM4I0P+x6wlwREk9MoSukl2pRyS6AwL+f4S6riySxxkH0lENQ/Dwsg4hjODo6XAhTYDW5z5iyLJrGlgDOon73ZCQrAnAdxfpIndj1g3+SRtVmG0PNxMA1y0itGfnltgbDeXTXhyRmr+GsQNe/vNQ==!!Dhoeo7t00/HT9DrsOU0u6IjrmXnlNvfnoYGIxESx1is1MfyS5D/qjE6m/12ARdZF4+Gy5s/gV0gwtF0e+0PgotMv7eH18Ozb+bQpoeNSepM6/PX4YiOovFa7+RujJmSzqlsCOkBTZd28WMCS1128VRL/UQFQ4LXnKHg7GmQtZLxr8rQQQuNgecUzGlkK1U/qqAssBCbmLX5cgemUrfwDFDfqwN5LiuYmgBGymfg2YFyeNYcSmsOem848WaW5FeWvoyKi24pLc098jhwE8NQvUz0XZay5Hk9jCln2dGmLsbYjNHsS5Z8NDN/Zj5k6uzT9oqWdmaj/Fd0l5+8TYnQ211D9JZ4Qx8qV1BKZjhVRx/CVvI1C784gpLO5SccAMtcmY3YM3DoWQPbLbQFcEEEen7545MDakM1NdPEkO4iUOKLBfMnKgIStW7IKmtNUB1INtmwbung7aE5aVRohEskatHLVAtJ0Rt+Lfiv4eF5zxo5G1m47KKa2rM4ovYA8QQPCCULThkbIDA168StpzKa8p6YeqjG1n/m9ie2UNBYJTQZ96AFm+VXX2VyeeiUSM2FPWwSo8BtBBNYKiFe0AO3HXx/HxFW/6YZWO96c65NWx8cYCfzaL3oBkY+wI3dujx1/jC0DaqZ/ALRLAad9sKi6Mk+nLuKlgVWDEc84CIe2KuC36qIJyHI+a+5XUm5Kg5H0zE0nXV4O59qXiL9xwpphtjMeo9A0cbHMo5yy14ajmqLnRVPhVaUMOz6Z2fhU/YSRI7076fanze8wxmPTSIqFr7r/t3M01D6u02+myTT2o5HX6jPsDsqhgTJeBTvS+yFWUk8ETEFx0o/DrzR7GnSgQd/8872h+oiu98iRbzpOBYlO2v95JMLWZIVPmlL6sd0MSm7+RwdBM7rHMTIy31YTpg==^&#
[2024-01-25] [10:53:15] [INFO] - Headers: #&^u2V88RBjxOXSU7roD/JBFDtMI9690u6p2zYKruQLbWDdw76Am9man7Y9WRPDgCRQx1nBj4ugxH4c6xpTd1lX7ZTaLAZab/qCHFNp3PZr8x75zUDF0KyFDGr689GSMqIYc8meBrewK7aj8T+eTsfnBFMyZj8qck+IBBU0+LyQHlW4iQAkBKg6qrLyP5rnLGQ51QxM4I0P+x6wlwREk9MoSukl2pRyS6AwL+f4S6riySxxkH0lENQ/Dwsg4hjODo6XAhTYDW5z5iyLJrGlgDOon73ZCQrAnAdxfpIndj1g3+SRtVmG0PNxMA1y0itGfnltgbDeXTXhyRmr+GsQNe/vNQ==!!5u/PW84zNpUNPOxuXg5r9I0klPPOCK1GkwiR6DYsYsWhEQlLBsvLaLxD9w7MIubZHLHreF1qgYmZ27OQW4oLSVE73iZdeEJTUIAnxv1SRJ69PstSfE+emBrjKoh4AaRzRvbGIXNn0gzXNJUOG+vydKkN2XPmG0xyeRVy0RC6Pbj98G0v0CE7aNUv3sLcvY6qxScVpxgArku5yHRnwPzn+BPpcaZHmkcQZqDQGPs3Fv2OkufxV4112YDuHOZZAAtPzEDN1W2PLcaP0dP2bWMWWl4LHu/S4Gi50ST+uMyv+InPurXEWcbYMpTedIrYfSemV1xjRQPWj1fcZ6EMMlhAFs2t9r2U/QVkT/K4Xm7IBmilC//CJ0MyJCWxyP7NHLhhsMLxN5idEpzWiN9X87uldthA/CqkW+bNkYa3lMqHQbMWVI8Woh5nR4pfWmkFpmEcpAQ48ZPpvRLLjqtMfNLWbBRE0ya8iufG21qAv28KErzKoOPWXFCgl5qynAlps57M/1G0N4C+r+u6CFb8ULpLElcpoVhZyDXEL8LzCRnYVhc=^&#
PHP Warning:  Trying to access array offset on null in D:\php\Bypass\bypass.php on line 355
[2024-01-25] [10:53:15] [INFO] - Sending POST request...
[2024-01-25] [10:53:15] [INFO] ! Fail to send request, check your internet connection.

D:\php>

NVG-064, 2024-01-25:
Where:
$init = decryptData($args);

and:
$data = json_decode(decryptData($args), true);

Thanks

MlgmXyysd, 2024-01-25:
> [2024-01-25] [10:53:15] [INFO] - Headers: **#&^**

https://twitter.com/realMlgmXyysd/status/1748386904050110675

closed 2024-02-19 by MlgmXyysd