MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #46: The system is being upgraded. Please try again later. (30003) [Redmi Note 13 Pro Plus (zircon)]

encouragerv, 2024-01-30:
I do all steps in "How to use". After binding i got this error (watch screen from PC and Phone). Trying to use MiFlashUnlock after error - nothing happen. Maybe this phone isn't supporting this tool because new SoC?

[image]
[image]

MlgmXyysd, 2024-01-31:
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/37

closed 2024-01-31 by MlgmXyysd

ninjadev64, 2024-06-13:
Hey @encouragerv

I have the same device, did you manage to solve this?