MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #53: Error reporting consultation

YunYe-C-X, 2024-02-05:
What is the reason for the following problem? How to solve it, the device is Xiaomi 13
[image]

stevleibelt, 2024-02-05:
@YunYe-C-X 

What do you get, when you run `php -i`?
Do you see that openssl is installed?
Maybe you have something like grep, than `php -i | grep -i openssl` is the way to go.

The error message is telling you, that it tries to call/execute the command `openssl_decrypt`but the command is not available.

Best regards.

YunYe-C-X, 2024-02-10:
I have installed openssl and curl using `php -i openssl `and `php -i curl`. I still get the same error when I run it again. What is the reason for this?

YunYe-C-X, 2024-02-10:
[image]

YunYe-C-X, 2024-02-10:
[image]

YunYe-C-X, 2024-02-10:
@stevleibelt Please help me figure out how to proceed with this. Think you~

YunYe-C-X, 2024-02-10:
Thank you. The problem has been solved. It is a problem with the php.ini file configuration.

stevleibelt, 2024-02-10:
@YunYe-C-X perfect, good to read. Indeed I was not able to help since I can use Linux or FreeBSD but not Windows.

Best of luck and if you could explain the problem plus a solution, some else will send you good karma in the future!

closed 2024-02-19 by MlgmXyysd