MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #57: Got error (10000)

andrewlbs, 2024-02-14:
Hi,
I have a Xiaomi Redmi Note 13 Pro + with a Chinese phone.
I updated to XyperOS and tried to make BootLoader unlock with Bypass, but I received error 10000

[2024-02-14] [14:16:25] [INFO] * Now you can bind account in the developer options.
[2024-02-14] [14:16:47] [INFO] - Account bind request found! Let's block it.
[2024-02-14] [14:16:47] [INFO] - Refactoring parameters...
PHP Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333
[2024-02-14] [14:16:47] [INFO] - Sending POST request...
[2024-02-14] [14:16:49] [INFO] - Request parameter **error (10000)**

Could someone assist me, please? I am unsure of the meaning of this error message.

AbdElrahmanShaker446, 2024-02-19:
i have the same issue did u find a fix ?

MWTJC, 2024-02-19:
From phone model (Redmi Note 13*): [#37 ](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/37#issuecomment-1913903306)  (seems means official blocking)
From error number (10000): https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/54#issuecomment-1933645502 (seems means os version too high)

MlgmXyysd, 2024-02-19:
Redmi Note 13 series doesn't allowed to unlock BootLoader #37 
https://twitter.com/realMlgmXyysd/status/1748386904050110675

closed 2024-02-19 by MlgmXyysd