MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #60: 这是什么

vicdorke, 2024-02-20:
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in D:\a\bypass.php:230
Stack trace:
#0 D:\a\bypass.php(330): decryptData('#&^rcllVjPM614M...')
#1 {main}
  thrown in D:\a\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in D:\a\bypass.php:230
Stack trace:
#0 D:\a\bypass.php(330): decryptData('#&^rcllVjPM614M...')
#1 {main}
  thrown in D:\a\bypass.php on line 230

fisker, 2024-02-20:
Add `extension=ext/php_openssl.dll` to your `php.ini`

vicdorke, 2024-02-20:
替换setting失败，k50pro.提供的setting闪退，不兼容

Dalabomba, 2024-02-20:
> Add `extension=ext/php_openssl.dll` to your `php.ini`

I did so as you mentioned. But I still get the following report:

PHP Fatal error:  Uncaught Error: Call to undefined function curl_init() in D:\IDM Download\Bypass-1.0-fix-universal\bypass.php:158
Stack trace:
#0 D:\IDM Download\Bypass-1.0-fix-universal\bypass.php(199): http('https://unlock....', 'POST', 'data=%7B%22clou...', Array, true)
#1 D:\IDM Download\Bypass-1.0-fix-universal\bypass.php(345): postApi('unlock/applyBin...', Array, Array, true)
#2 {main}
  thrown in D:\IDM Download\Bypass-1.0-fix-universal\bypass.php on line 158

Fatal error: Uncaught Error: Call to undefined function curl_init() in D:\IDM Download\Bypass-1.0-fix-universal\bypass.php:158
Stack trace:
#0 D:\IDM Download\Bypass-1.0-fix-universal\bypass.php(199): http('https://unlock....', 'POST', 'data=%7B%22clou...', Array, true)
#1 D:\IDM Download\Bypass-1.0-fix-universal\bypass.php(345): postApi('unlock/applyBin...', Array, Array, true)
#2 {main}
  thrown in D:\IDM Download\Bypass-1.0-fix-universal\bypass.php on line 158

MlgmXyysd, 2024-02-22:
> 替换setting失败，k50pro.提供的setting闪退，不兼容

自己提取

Tiquillo, 2024-02-22:
I had the same error

```
[2024-02-22] [19:54:14] [INFO] - Starting ADB server...
[2024-02-22] [19:54:15] [INFO] - Processing device XSQG6DUWWOS4FY5T(4)...
[2024-02-22] [19:54:15] [INFO] - Finding BootLoader unlock bind request...
[2024-02-22] [19:54:16] [INFO] * Now you can bind account in the developer options.
[2024-02-22] [19:55:07] [INFO] - Account bind request found! Let's block it.
[2024-02-22] [19:55:07] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in C:\a\bypass.php:230
Stack trace:
#0 C:\a\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\a\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\a\bypass.php:230
Stack trace:
#0 C:\a\bypass.php(330): decryptData('8/8Mnp7WcRph8sb...')
#1 {main}
  thrown in C:\a\bypass.php on line 230
```
And going deeper in the error `Call to undefined function openssl_decrypt()` I found that I should add `extension=php_openssl.dll` to the `php.ini` file below line 950 (below `;extension=shmop`) and also uncomment the line 768 (remove `;` from `;extension_dir = "ext"`) to tell php where are the extensions located.

Finally, I got the following exit:

```
[2024-02-22] [19:57:15] [INFO] - Starting ADB server...
[2024-02-22] [19:57:15] [INFO] - Processing device XSQG6DUWWOS4FY5T(4)...
[2024-02-22] [19:57:16] [INFO] - Finding BootLoader unlock bind request...
[2024-02-22] [19:57:16] [INFO] * Now you can bind account in the developer options.
[2024-02-22] [19:57:28] [INFO] - Account bind request found! Let's block it.
[2024-02-22] [19:57:28] [INFO] - Refactoring parameters...
[2024-02-22] [19:57:28] [INFO] - Sending POST request...
[2024-02-22] [19:57:31] [INFO] - Target account: 6595687508
[2024-02-22] [19:57:31] [INFO] - Account bound successfully, wait time can be viewed in the unlock tool.
```

vicdorke, 2024-02-25:
> I had the same error
>
> ```
> [2024-02-22] [19:54:14] [INFO] - Starting ADB server...
> [2024-02-22] [19:54:15] [INFO] - Processing device XSQG6DUWWOS4FY5T(4)...
> [2024-02-22] [19:54:15] [INFO] - Finding [...]

https://www.123pan.com/s/625SVv-TGxbh.html
替换setting  apk