MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #61: 出现这个报错，怎么处理

LanSeDiQiu, 2024-02-20:
[image]

**小米解锁程序显示已经安装了驱动**

[image]

vicdorke, 2024-02-21:
替换setting.apk

shoshi007, 2024-02-22:
> 替换setting.apk

can you help ?

nttg430, 2024-02-22:
Has anyone had success with Redmi Note 13 Pro-Rom China?

shoshi007, 2024-02-22:
> Has anyone had success with Redmi Note 13 Pro-Rom China?
no

LanSeDiQiu, 2024-02-22:
> 替换setting.apk

下载的卡包解包后都是.img文件，但是总是错误，解包不出来system_ext文件，是卡包不对吗

ubun222, 2024-02-25:
aes解密fail了

vicdorke, 2024-02-25:
> > 替换setting.apk
> 
> can you help ?

https://www.123pan.com/s/625SVv-TGxbh.html

Erixrez, 2024-02-26:
I too have the same problem. Before this the script was running fine. For now i manage to solve by using MiCommunity apk by changing region to **Global** and now you can apply for unlocking.

asklepios73, 2024-02-28:
> I too have the same problem. Before this the script was running fine. For now i manage to solve by using MiCommunity apk by changing region to **Global** and now you can apply for unlocking.

It doen't work in China

Erixrez, 2024-02-28:
> > I too have the same problem. Before this the script was running fine. For now i manage to solve by using MiCommunity apk by changing region to **Global** and now you can apply for unlocking.
> 
> It doen't work in China

Unninstall the MiCommunity pre built version and download from google.

asklepios73, 2024-02-28:
Thats what I did. I have been granted access to unlock until next year with MiCommunity, but still cannot add acount and device, 
[image]
It seems the code in line 333 doesn't work:
$data["rom_version"] = str_replace("V816", "V14", $data["rom_version"]);

SergChe88, 2024-03-07:
Open bypass.php in any editor (notepad++ for example), and insert string $data["rom_version"] = ""; before line 333
[image]

asklepios73, 2024-03-07:
It doesn't work:
[2024-03-07] [14:05:24] [INFO] - The system is being upgraded. Please try again later. (30003)

SergChe88, 2024-03-07:
> It doesn't work: [2024-03-07] [14:05:24] [INFO] - The system is being upgraded. Please try again later. (30003)

Yes, me too. 
Can't bind phone to account.

MlgmXyysd, 2024-03-07:
Duplicated #37

closed 2024-03-07 by MlgmXyysd