MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #65: K70P一直显示Now you can bind account in the developer options.

RusanNg, 2024-02-29:
[image]
手机弹出 开发者选项 后，没有显示需要绑定账号的弹框，提示，也没有其它反应。

0874hc, 2024-03-06:
请问现在解决了吗

MlgmXyysd, 2024-03-07:
看报错你似乎没配好依赖项

0874hc, 2024-03-07:
请问这个是可以直接解锁k70bl是嘛

RusanNg, 2024-03-13:
> 看报错你似乎没配好依赖项

配置好了，我的机型是 k70p，出厂是 澎湃 os，我看了其它网友的帖子，好像新机型，出厂是澎湃 os 的，就没法解锁。

yezige, 2024-03-14:
这不是报错，这个是需要去开发者选项里找到 `设备解锁状态`，然后点击`绑定账号和设备`

RusanNg, 2024-03-14:
但是在开发者选项没有找到 设备解锁状态 这个选项…

yezige, 2024-03-15:
> 但是在开发者选项没有找到 设备解锁状态 这个选项…
> […](#)

有的，看下图
[image]

yezige, 2024-03-15:
不过建议你k70p不用试了，我k70昨天刚试的，后边会报30001错误

chenhao023, 2024-03-23:
xiaomi14也一样，30001错误

AlaskaSier, 2024-03-25:
出廠澎湃的都不可以吧

AlaskaSier, 2024-03-25:
安安靜靜等待大佬出新的破解

zdd530, 2024-03-27:
耐心等待出出厂是澎湃的破解方法

Tigercat000, 2024-04-24:
剛試K70 Pro 都是出30001，應該暫時沒解了

xbn2002, 2024-05-05:
> 耐心等待出出厂是澎湃的破解方法

不用等了，目前没有好的方法，不如先答题签到

zhbcher, 2024-05-23:
签到答题，我晕，就连雷子都才四级，让你达到五级，这不是逗你玩的吗？

yezige, 2024-05-23:
铁子们，我快五级了，EU版的系统还能用指纹支付的功能吗？