MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #66: Request parameter error (10000)

KreitinnSoftware, 2024-02-29:
I tried to use in my (Poco X5 Pro 8GB 256GB India) and got Request parameter error (10000), I tried on Linux and Windows, my Xiaomi Account is valid and the device it's unbanned

[pablo@localhost Xiaomi-HyperOS-BootLoader-Bypass]$ php bypass.php 
[2024-02-29] [21:07:24] [INFO] - ************************************
[2024-02-29] [21:07:24] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-02-29] [21:07:24] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-02-29] [21:07:24] [INFO] - ************************************
[2024-02-29] [21:07:24] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-02-29] [21:07:24] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-02-29] [21:07:24] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-02-29] [21:07:24] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-02-29] [21:07:24] [INFO] - My Blog: https://www.neko.ink/
[2024-02-29] [21:07:24] [INFO] - ************************************
[2024-02-29] [21:07:24] [INFO] - Starting ADB server...
[2024-02-29] [21:07:24] [INFO] - Processing device 38c76b78(device:redwoodin)...
[2024-02-29] [21:07:24] [INFO] - Finding BootLoader unlock bind request...
[2024-02-29] [21:07:25] [INFO] * Now you can bind account in the developer options.
[2024-02-29] [21:07:28] [INFO] - Account bind request found! Let's block it.
[2024-02-29] [21:07:28] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in /home/pablo/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 333
[2024-02-29] [21:07:28] [INFO] - Sending POST request...
[2024-02-29] [21:07:30] [INFO] - Request parameter error (10000)

GeorgeSakketos, 2024-03-03:
I have the same exact problem with the Redmi Pad (both when connected to Mobile Data through a hotspot on my phone and when it's connected to my ISP wifi)

water2152, 2024-03-04:
I have the same issue