MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #69: Request parameter error (10000)

LongZijie, 2024-03-05:
Account bind request found! Let's block it.
[2024-03-05] [14:41:45] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in C:\PHP\bypass.php on line 333

Warning: Trying to access array offset on null in C:\PHP\bypass.php on line 333
PHP Deprecated:  str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in C:\PHP\bypass.php on line 333

Deprecated: str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in C:\PHP\bypass.php on line 333
[2024-03-05] [14:41:45] [INFO] - Sending POST request...
[2024-03-05] [14:41:47] [INFO] - Request parameter error (10000)

can pls help?

LongZijie, 2024-03-06:
[image]

LongZijie, 2024-03-06:
i am using a 13u cn version, so i should install this settings APK, run the script again, and uninstall the update after bypass, correct? thanks
[image]

LongZijie, 2024-03-06:
[image]
this means the bypass works alrdy right?

LongZijie, 2024-03-07:
[image]

LongZijie, 2024-03-07:
seems it worked. thank u

Temeramor, 2024-03-07:
Really worked?
What did you do for this?

closed 2024-03-07 by MlgmXyysd

LongZijie, 2024-03-08:
> Really worked? What did you do for this?

install the Setting.apk here [Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip)

Patator83, 2026-11-09:
Hello,
Same message but for a redmi note 15 pro lapis. Do I also have to install a setting.apk ? How to create one?