MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #71: 卡在Now you can bind account in the developer options.

bsns, 2024-03-08:
分别测试按照redme和网友分享的整合包，均停留在卡在Now you can bind account in the developer options.
[image]
手机端点击“绑定账号和设备”时，迅速提示失败，同时电脑端无反应；
设备为小米13，版本号为Hyper OS1.0.24.2.26.DEV

xbn2002, 2024-05-05:
现在好了吗？这种情况应该是电脑一直没能捕获到绑定请求导致，手机那边已经发出去了，电脑这里没有截获
考虑是adb驱动的问题，或者数据线问题，或者你手机网络太快了，改成3G试试

Linux-K-Git, 2024-06-08:
同这个问题，K60Ultra，也是卡在这一步，手机上无论更换账户还是换卡改网重启都不行，电脑这边始终无法截获

Linux-K-Git, 2024-06-08:
确定adb驱动是正常的，可以看到adb授权

shanlanCoding, 2024-09-09:
我贴一下我的报错，之前没有解锁过，手机型号小米13，刚更新的澎湃`V816.0.13.0.UMCCNXM`，公告版本名称`1.0.13.0.UMCCNXM`
[image]

beginningof, 2024-09-29:
[image]
把前面的分号去掉后，手机端点击“绑定账号和设备”时，还是提示失败

chiyuelao, 2024-10-31:
一样问题。。

zhiyuanisme, 2024-11-22:
跟你一样的问题  我点进去《设备解锁状态》显示的是一个图标，根本没有输入框