MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #74: Developer options closes with settings.apk replaced

kingster85, 2024-03-09:
On Redmi Pad I was able to replace settings apk and bind my device.
On my Redmi Note 13 Pro+ however, the developer options screen immediately closes.
Whenever I force to open the "Mi Unlock Status" screen I get an java error (something like security exceptions blabla).
Any fix for this?
Is it a prerequisite to remove any security or factory reset without setting up security?
Let me know.

xbn2002, 2024-03-12:
This product series is not supported to unlock. There is no offical way.

florezriascos, 2024-03-25:
The same thing happened to me, I solved it by turning off the bluetooth....
try'it

Kimarnic, 2024-09-17:
> The same thing happened to me, I solved it by turning off the bluetooth.... try'it

You saved me, I love you, It kept crashing and crashing, but disabling bluetooth fixed it, I have to wait 168 hours (a week) now.
I love you