MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #76: The binding is failed with toast '验证失败，请稍后再试' even after the modified request is success,

lonelam, 2024-03-10:
Hi, my phone model is Xiaomi 13 Ultra and it is uploaded to Hyper OS 1.0.6.0 .
My first try with the script failed at the decryption step, so I searched and found a solution to replace Settings with the downgrade apk.
After I installed the Settings.apk from #29 , the request is successfully sent and I see the response with a "code": 0, so all the steps in the script seems fine, but the status in '设备解锁状态' is still unchanged, and the fastboot mode phone can still not be recognized by my PC (the list is empty for `./libraries/fastboot.exe devices` )

[image]

lonelam, 2024-03-10:
I printed the result, and I think the request is successfully sent, but what's the problem with the fastboot mode? It seems the driver did not recognize the Xiaomi 13 Ultra phone.

```
$ "C:\Software\php-8.3.3-Win32-vs16-x64\php.exe" bypass.php 
[2024-03-10] [22:23:19] [INFO] - ************************************
[2024-03-10] [22:23:19] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-03-10] [22:23:19] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-03-10] [22:23:19] [INFO] - ************************************
[2024-03-10] [22:23:19] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-03-10] [22:23:19] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-03-10] [22:23:19] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-03-10] [22:23:19] [INFO] - My Blog: https://www.neko.ink/
[2024-03-10] [22:23:19] [INFO] - ************************************
[2024-03-10] [22:23:19] [INFO] - Starting ADB server...
[2024-03-10] [22:23:20] [INFO] - Processing device b48c39xx(1)...
[2024-03-10] [22:23:20] [INFO] - Finding BootLoader unlock bind request...
[2024-03-10] [22:23:20] [INFO] * Now you can bind account in the developer options.
[2024-03-10] [22:23:20] [INFO] - Account bind request found! Let's block it.
[2024-03-10] [22:23:20] [INFO] - Refactoring parameters...
[2024-03-10] [22:23:20] [INFO] - Sending POST request...
[2024-03-10] [22:23:21] [INFO] - {"result":"ok","retriable":false,"code":0,"data":{"userId":"16454934xx","status":"LOCKED"},"description":"\u6210\u529f","ts":1710109401516}
[2024-03-10] [22:23:21] [INFO] - Target account: 1645493450
[2024-03-10] [22:23:21] [INFO] - Account bound successfully, wait time can be viewed in the unlock tool.
```

MlgmXyysd, 2024-03-11:
> Q: The device shows Couldn't verify, wait a minute or two and try again.
> A: This is normal, the binding request on the device side has been blocked by our script. The actual binding result is subject to the script prompt.

closed 2024-03-11 by MlgmXyysd