MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #77: Unlock phone with hyperos china

pjakubina, 2024-03-11:
When I run the bypass.php script with the phone connected with debug mode I get this error in the console
```
PHP Fatal error:  Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /root/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 330 and defined in /root/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php:228
Stack trace:
#0 /root/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php(330): decryptData()
#1 {main}
  thrown in /root/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 228
```

How to fix it

mingyougei, 2024-03-16:
你手机是redmi note 13系列的吗
是的话解不了

xbn2002, 2024-05-05:
It seems to be a problem with the php environment, check it