MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #82: PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt()

Keronaya, 2024-03-18:
Phone type: Mi 12s ultra

`PHP Warning:  PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (The specified module could not be found), C:\php\ext\php_curl.dll (The specified module could not be found)) in Unknown on line 0

Warning: PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (The specified module could not be found), C:\php\ext\php_curl.dll (The specified module could not be found)) in Unknown on line 0
PHP Warning:  PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (The specified module could not be found), C:\php\ext\php_openssl.dll (The specified module could not be found)) in Unknown on line 0

Warning: PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (The specified module could not be found), C:\php\ext\php_openssl.dll (The specified module could not be found)) in Unknown on line 0
[2024-03-18] [03:44:11] [INFO] - ************************************
[2024-03-18] [03:44:11] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-03-18] [03:44:11] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-03-18] [03:44:11] [INFO] - ************************************
[2024-03-18] [03:44:11] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-03-18] [03:44:11] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-03-18] [03:44:11] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-03-18] [03:44:11] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-03-18] [03:44:11] [INFO] - My Blog: https://www.neko.ink/
[2024-03-18] [03:44:11] [INFO] - ************************************
[2024-03-18] [03:44:11] [INFO] - Starting ADB server...
[2024-03-18] [03:44:12] [INFO] - Processing device eb8ae706(1)...
[2024-03-18] [03:44:12] [INFO] - Finding BootLoader unlock bind request...
[2024-03-18] [03:44:12] [INFO] * Now you can bind account in the developer options.
[2024-03-18] [03:44:16] [INFO] - Account bind request found! Let's block it.
[2024-03-18] [03:44:16] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\jc973518\OneDrive - James Cook University\Bypass-1.0-fix-universal\bypass.php:230
Stack trace:
#0 C:\Users\xxx\OneDrive - James Cook University\Bypass-1.0-fix-universal\bypass.php(330): decryptData('#&^LCB7t0rQ+RK/...')
#1 {main}
  thrown in C:\Users\xxx\OneDrive - James Cook University\Bypass-1.0-fix-universal\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\jc973518\OneDrive - James Cook University\Bypass-1.0-fix-universal\bypass.php:230
Stack trace:
#0 C:\Users\xxx\OneDrive - James Cook University\Bypass-1.0-fix-universal\bypass.php(330): decryptData('#&^LCB7t0rQ+RK/...')
#1 {main}
  thrown in C:\Users\xxx\OneDrive - James Cook University\Bypass-1.0-fix-universal\bypass.php on line 230`

Keronaya, 2024-03-18:
According to the video, I final solve it.
https://www.bilibili.com/video/BV1me411q7wZ/?spm_id_from=333.1007.top_right_bar_window_history.content.click&vd_source=984e922e999474896ae171bd4d0ef4d6

[image]

download the setting.apk then install on your phone, it will be work for you!

cikgit, 2024-03-21:
> According to the video, I final solve it. https://www.bilibili.com/video/BV1me411q7wZ/?spm_id_from=333.1007.top_right_bar_window_history.content.click&vd_source=984e922e999474896ae171bd4d0ef4d6
> 
> ![1710735412747](https://private-user-images.githubusercontent.com/19572461/313548569-71758ee4-c9b6-40d8-be1d-716b9c5c71ff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwMzQxMTEsIm5iZiI6MTcxMTAzMzgxMSwicGF0aCI6Ii8xOTU3MjQ2MS8zMTM1NDg1NjktNzE3NThlZTQtYzliNi00MGQ4LWJlMWQtNzE2YjljNWM3MWZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDE1MTAxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJjMTY3ZTVhNjEyM2EzOTcwYTMzMmMyZWUyMmVlNTJjMzc5ZTBmOTQyYmM4ZGRhNzNkMDcwNzViZmY3OGZiOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Xz4DgqfH6yiVTTVlQnB-CYbHICZ_NHp5NilAiaiPBVA)
> 
> download the setting.apk then install on your phone, it will be work for you!

can you please upload that files to mega or gdrive? most of people cant download from chinese file servers! **they require chinese mobile number to sign up for downloading.**

son54188, 2024-03-21:
hey can u help me looking for setting file of xiaomi 13 pro UMBCNXM, i have same proglem with u

Keronaya, 2024-03-22:
https://1drv.ms/u/s!AnILV8pyuEEI21pIYPZNjdyJJXqY?e=6Mydyz

This setting.apk is not specific for xiaomi 13 pro or another but it can be work!

@son54188 @cikgit

cikgit, 2024-03-23:
> https://1drv.ms/u/s!AnILV8pyuEEI21pIYPZNjdyJJXqY?e=6Mydyz
> 
> This setting.apk is not specific for xiaomi 13 pro or another but it can be work!
> 
> @son54188 @cikgit

Thanks for your reply.
My device Note 13 Prp+ CN
This settings.apk same as mine (according to app manager)
System doesn't let me install it, because it's from unknown channel
[image]

I think only way to unlock it buy UBL token from techie

LeanFly, 2024-03-27:
https://www.123pan.com/s/625SVv-BYxbh.html

从上面网盘下载setting.apk，安装后重启手机

Tigercat000, 2024-04-24:
> https://www.123pan.com/s/625SVv-BYxbh.html
> 
> 从上面网盘下载setting.apk，安装后重启手机

請問有K70PRO的apk嗎? Thanks

magicduke, 2024-05-06:
> According to the video, I final solve it. https://www.bilibili.com/video/BV1me411q7wZ/?spm_id_from=333.1007.top_right_bar_window_history.content.click&vd_source=984e922e999474896ae171bd4d0ef4d6 [...]

大哥，不说了，给你磕一个

theincognito-inc, 2024-05-07:
@Keronaya Is it also possible for you to download and upload the Xiaomi Pad 6 series settings apk from https://www.123pan.com/s/625SVv-BYxbh.html ? 🙏🏼 It needs chinese phone number and I unfortunately don't have that.

liuweigeek, 2024-05-16:
> @Keronaya Is it also possible for you to download and upload the Xiaomi Pad 6 series settings apk from https://www.123pan.com/s/625SVv-BYxbh.html ? 🙏🏼 It needs chinese phone number and I unfortunately don't have that.

@theincognito-inc I grabbed the APK for the Xiaomi Pad 6 Series from the link and uploaded it to Google Drive, please find it in https://drive.google.com/file/d/1kofHDgk-2Jsd-hLKZ6nP2H5gCk9yfNX7/view?usp=sharing

theincognito-inc, 2024-05-16:
> @theincognito-inc I grabbed the APK for the Xiaomi Pad 6 Series from the link and uploaded it to Google Drive, please find it in https://drive.google.com/file/d/1kofHDgk-2Jsd-hLKZ6nP2H5gCk9yfNX7/view?usp=sharing

This worked and I got a 168h timer for my Pad 6S Pro. Thank you!

zimmistalin, 2024-05-27:
big thanks for ur work, im happy my bootloader is unlock now cuz of you.

cbmland, 2024-05-28:
> According to the video, I final solve it. https://www.bilibili.com/video/BV1me411q7wZ/?spm_id_from=333.1007.top_right_bar_window_history.content.click&vd_source=984e922e999474896ae171bd4d0ef4d6 [...]

[image]
[image]

its work on my redmi turbo 3, and checked by unlock tools, showing waiting 168 hours!

akasam2231, 2024-06-03:
hi there its not working for me i get the same error 1000. i have a mi 10 pro on hyper os can i have the settings apk or whatever i need to get rid of this error

RezzZ, 2024-06-04:
> > According to the video, I final solve it. https://www.bilibili.com/video/BV1me411q7wZ/?spm_id_from=333.1007.top_right_bar_window_history.content.click&vd_source=984e922e999474896ae171bd4d0ef4d6
> > ![1710735412747](https://private-user-images.githubusercontent.com/19572461/313548569-71758ee4-c9b6-40d8-be1d-716b9c5c71ff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5MDMwMDQsIm5iZiI6MTcxNjkwMjcwNCwicGF0aCI6Ii8xOTU3MjQ2MS8zMTM1NDg1NjktNzE3NThlZTQtYzliNi00MGQ4LWJlMWQtNzE2YjljNWM3MWZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDEzMjUwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMwOWVjZjYzOWU1YmFiYjE5NGM2YTk3ZWM0MzQ4ODE0YjY2NTI1NzY5NWFmNGQ2NmU0NTdkMzZkYjM1NDU5ZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xRYbq4PNZO1s2g5QBu8zLFfZsI6ka36uF2J5X9A4avM)
> > download the setting.apk then install on your phone, it will be work for you!
> 
> <img alt="Snipaste_2024-05-28_21-16-50" width="1169" src="https://private-user-images.githubusercontent.com/660116/334429832-6d685f0a-64c4-43d6-81d9-dc246a684232.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1MDI0NzEsIm5iZiI6MTcxNzUwMjE3MSwicGF0aCI6Ii82NjAxMTYvMzM0NDI5ODMyLTZkNjg1ZjBhLTY0YzQtNDNkNi04MWQ5LWRjMjQ2YTY4NDIzMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNFQxMTU2MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02Y2I0YzU3OTk3OWNiNDc2OTg0ZTRiMjRjMGUwMDQ2YWQ3MGIxMzliOTZiMDc5ZmExZGQzMjc0Mjg1MDI5MTU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gXJXExNQ6_JHE_foS9yKRuhjfiGos8zo5RwUmaM_0vQ"> <img alt="Snipaste_2024-05-28_21-15-35" width="922" src="https://private-user-images.githubusercontent.com/660116/334429903-327d7623-c424-48c3-956d-3e069a0b1c4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1MDI0NzEsIm5iZiI6MTcxNzUwMjE3MSwicGF0aCI6Ii82NjAxMTYvMzM0NDI5OTAzLTMyN2Q3NjIzLWM0MjQtNDhjMy05NTZkLTNlMDY5YTBiMWM0ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNFQxMTU2MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MWM3NGU1NDQxNWFlZGRiODY2MTIxYzJmZGY1ZjQ4ZmM3OTU5OTg4YTZiZjY4YWNkZjY5NTlkODQ3ODk0ZjI2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.b83KTXYDCkX_WJptjmIUO7-JtBcQ7ybnmrfgWYH8Hxk">
> its work on my redmi turbo 3, and checked by unlock tools, showing waiting 168 hours!

I also got the account bound successfully message but every time I try to boot in fastboot, the phone restarts again and boots hyperos instead. Any tips?

ZaryabKhan, 2024-06-11:
> @theincognito-inc I grabbed the APK for the Xiaomi Pad 6 Series from the link and uploaded it to Google Drive, please find it in [...]

It worked in my Redmi 12 (fire) as well. Thank you so much! Now I have to wait 168 hours 🤞

z-ww447, 2024-06-25:
Have to wait 168 hours. Can I just leave my phone in fastboot mode plugged into my PC and wait, or do I have to have it on with my SIM?

ZaryabKhan, 2024-06-25:
> Have to wait 168 hours. Can I just leave my phone in fastboot mode plugged into my PC and wait, or do I have to have it on with my SIM?

Just use it normally, like you were using before applying for the bootloader unlock.

z-ww447, 2024-06-25:
> > Have to wait 168 hours. Can I just leave my phone in fastboot mode plugged into my PC and wait, or do I have to have it on with my SIM?
> 
> Just use it normally, like you were using before applying for the bootloader unlock.

I'm currently on another phone, don't really want to switch if I have to wipe my data after unlocking

ZaryabKhan, 2024-06-25:
> I'm currently on another phone, don't really want to switch if I have to wipe my data after unlocking

Okay, it's up to you if you don't want to use it while waiting but I suggest you to use it at least 1 hour daily and turn on Find my phone in mi account settings...i heard that keeping your phone idle for 168 hours will increase the unlock time to 15 days.

z-ww447, 2024-06-25:
> Okay, it's up to you if you don't want to use it while waiting but I suggest you to use it at least 1 hour daily and turn on Find my phone in mi account settings...i heard that keeping your phone [...]

Guess I have to carry two phones and use hotspot to connect to my main phone...
Just tried to unlock again to see if it has gone down from 168
Anddd it "couldn't verify device"

RezzZ, 2024-06-26:
My device was powered off for 95%+ of the time and didn't affect the wait time. But better safe than sorry? 😅

> Okay, it's up to you if you don't want to use it while waiting but I suggest you to use it at least 1 hour daily and turn on Find my phone in mi account settings...i heard that keeping your phone [...]

z-ww447, 2024-06-26:
> My device was powered off for 95%+ of the time and didn't affect the wait time. But better safe than sorry? 😅

Came home today, plugged my sim in, turned on my phone and after an hour tried to unlock. The time did go down, seems like it's working

RezzZ, 2024-06-26:
> Came home today, plugged my sim in, turned on my phone and after an hour tried to unlock. The time did go down, seems like it's working

I think even the sim is no longer required once the countdown has been started.

closed 2024-07-02 by MlgmXyysd

eddiemukri01, 2024-07-16:
How can i install that setting.apk? should I use adb install?

eddiemukri01, 2024-07-16:
> > @theincognito-inc I grabbed the APK for the Xiaomi Pad 6 Series from the link and uploaded it to Google Drive, please find it in https://drive.google.com/file/d/1kofHDgk-2Jsd-hLKZ6nP2H5gCk9yfNX7/view?usp=sharing
> 
> This worked and I got a 168h timer for my Pad 6S Pro. Thank you!

How can i install that setting.apk? should I use adb install?

NITIN911, 2024-08-03:
Is anybody have setting.apk for black shark 1?

Any help !! Anyone ?

blu3cryp70, 2024-09-23:
hey can u help me looking for setting file of mi 12 5G UMXINXM?
I looked into website but there is no file for mi 12 5G

Maulik-darji, 2024-09-24:
please please find get me setting apk for Redmi 13c 5g UGQINXM

Maulik-darji, 2024-09-24:
> UBL token

please help me to get setting.apk for Redmi 13c 5g indian mainland rom UGQINXM please bro

RezzZ, 2024-09-24:
@Maulik-darji did you try the Xiaomi Pad 6 Series settings.apk? it worked for my redmi turbo3

Smartslabs, 2024-09-24:
> @Maulik-darji did you try the Xiaomi Pad 6 Series settings.apk? it worked for my redmi turbo3

You can unlock turbo 3?

RezzZ, 2024-09-24:
yes

Smartslabs, 2024-09-24:
Sound great

Maulik-darji, 2024-09-24:
please guys help me to get the setting apk for redmi 13c 5g please please please

Smartslabs, 2024-09-24:
> @Maulik-darji did you try the Xiaomi Pad 6 Series settings.apk? it worked for my redmi turbo3

Xiaomi Pad 6 Series settings.apk work for turbo 3 ? Is still wait 168 hours?

Maulik-darji, 2024-09-24:
> > @Maulik-darji did you try the Xiaomi Pad 6 Series settings.apk? it worked for my redmi turbo3
> 
> Xiaomi Pad 6 Series settings.apk work for turbo 3 ? Is still wait 168 hours?

bro I am asking for Redmi 13c 5g. because settings apk for xiaomi pad 6 is not working on redmi 13c 5g

Smartslabs, 2024-09-24:
> yes
> […](#)

Xiaomi Pad 6 Series settings.apk work for turbo 3 ? Is still wait 168 hours?

Smartslabs, 2024-09-24:
@RezzZ 
Xiaomi Pad 6 Series settings.apk work for turbo 3 ? Is still wait 168 hours?

Maulik-darji, 2024-09-24:
are you mad ? I said it is not working for me

RezzZ, 2024-09-24:
> @RezzZ 
> 
> Xiaomi Pad 6 Series settings.apk work for turbo 3 ? Is still wait 168 hours?

Yes it worked with Chinese turbo 3

RezzZ, 2024-09-24:
> are you mad ? I said it is not working for me 

Dude chill. I mentioned it worked for my Chinese Xiaomi Redmi turbo 3, that's why I asked. You asking for another version doesn't mean I can smell it's not working for your device okay. Anyways, i can't help.i can only share what worked for me and my device

Maulik-darji, 2024-09-25:
> Dude chill. I mentioned it worked for my Chinese Xiaomi Redmi turbo 3, that's why I asked. You asking for another version doesn't mean I can smell it's not working for your device okay. Anyways, i [...]

sorry dude I am just replying to smartlabs

RezzZ, 2024-09-25:
> sorry dude I am just replying to smartlabs

and smartlabs was responding to me comment ;)

Smartslabs, 2024-09-25:
> > @RezzZ
> > Xiaomi Pad 6 Series settings.apk work for turbo 3 ? Is still wait 168 hours?
> 
> Yes it worked with Chinese turbo 3

Hey RezzZ, just wanted to check again—did you use the Xiaomi Pad 6 Series settings.apk to unlock Turbo 3, or did you use a specific settings.apk for Turbo 3?

RezzZ, 2024-09-25:
I used the pad6 APK

Smartslabs, 2024-09-25:
> I used the pad6 APK
> […](#)

Thanks for your reply! Since you had success, I’m interested in buying a Turbo 3 to try unlocking it myself.

kakalah, 2024-09-30:
> It worked in my Redmi 12 (fire) as well. Thank you so much! Now I have to wait 168 hours 🤞

After you unlock the device, does the app settings version change to the default version or is it still this version? If it is still this version, is it okay to delete this version when it is unlocked?

kakalah, 2024-09-30:
> @Maulik-darji did you try the Xiaomi Pad 6 Series settings.apk? it worked for my redmi turbo3

After you unlock the device, does the app settings version change to the default version or is it still this version? If it is still this version, is it okay to delete this version when it is unlocked?

RezzZ, 2024-09-30:
> > @Maulik-darji did you try the Xiaomi Pad 6 Series settings.apk? it worked for my redmi turbo3
> 
> After you unlock the device, does the app settings version change to the default version or is it still this version? If it is still this version, is it okay to delete this version when it is unlocked?

The idea is you flash the global or EU tom after unlocking so the settings app will be restored to the one provided in the room you choose

cmtrkr83, 2024-10-10:
redmi note 13 pro(garnet) there is no settings.apk. ..... not solved for rn13pro!

dmrangel, 2024-10-12:
can anyone help me find a settings.apk for my POCO X6 Pro?

preetam1409, 2024-10-17:
settings apk for poco f6 ?

juanjo123965, 2024-11-02:
alguien con la apk para el poco x6 igual tengo el mismo problema

bk60, 2024-11-02:
Xiaomi 14 Ultra same error   Request parameter error (10000) any solutions?

hungpham3112, 2024-11-16:
Hi everyone. Please have a look at my [comment](https://github.com/code4fun1024/setting-app-for-hyperos/issues/6#issuecomment-2480500936) to download the suitable settings apk for your ROM. Let me know if anything is missing

nandhu00090, 2024-11-25:
> can anyone help me find a settings.apk for my POCO X6 Pro?

did you find any solution?

Smartslabs, 2024-11-25:
If global version just get permission on Xiaomi committee apps at 00:00 every night is ok

preetam1409, 2024-12-05:
> > If global version just get permission on Xiaomi committee apps at 00:00 every night is ok
> 
> If it wasn't that simple we wouldn't be here... the reality is that for some devices and regions it has became at nightmare like the Poco F6 if you bought very recently.

actually it depends bcs I myself made 7 total account with  unlock permission on f6 global variant , also made few on a iqoo device

Mostwest, 2024-12-05:
> actually it depends bcs I myself made 7 total account with unlock permission on f6 global variant , also made few on a iqoo device

I don't doubt that. My account has been 400 days in Community App, tried everything: manual click, macrodroid, autoclickers. Nothing works still getting the same quota reached limit for the past 2 weeks since I started. Since I've been talking with multiple people who had the same problem It must be something region related.

What region did you choose when creating those 7 accounts?

[8 more replies]