MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #83: WAITING FOR DEVICE CONNECTION.

Lunikarchange, 2024-03-20:
I ran the script to bypass the bootloader restrictions and it launched perfectly. on the other hand, the only message that constantly scrolls just after STARTING ADB SERVER is: WAITING FOR DEVICE CONNECTION despite the fact that the phone is connected to the computer. I need help please.

[image]
[image]

TechFanTheo, 2024-03-23:
Same issue here

xbn2002, 2024-05-05:
Check if your platform-tools is working properly
And another way is to download a environment-intergrated version of this tool, you can search it on the internet.

kakazsh, 2024-05-18:
I have the same issue. The tool can't connect to the phone.
But running adb.exe works fine.
[image]

nitzcard, 2024-06-22:
same issue here, I see the phone in adb but script still saying this

MilesFarber, 2024-06-24:
Same issue. This is why people stick to iPhone.

jobrien2001, 2024-07-06:
Make sure you have adb in the right folder.  Check your task manager to see if adb.exe is running... if not... its not in the right folder.

In windows should be in PHPfolder\libraries

nNoidea, 2024-07-16:
got the same issue adb is running in the background, copy pasted the platform tools to `C` drive, the `php\libraries` folder and even the `php\lib` folder. None of them work.

KeksNino, 2025-01-04:
i have the same problem and i tried replacing libraries with the latest platform tools on linux, but it still doesnt work

2aq112x35, 2025-02-26:
> i have the same problem and i tried replacing libraries with the latest platform tools on linux, but it still doesnt work

download release.zip and copy files to 'libraries' folder.

sonofevil, 2025-03-10:
I think for Linux users this happens because the "libraries" directory simply does not exist. Linux installs are structured very differently from Windows installs where everything belonging to a program is in a single folder. There is no "php folder", the php binary is in `/usr/bin`, extensions are in '/usr/lib/php/modules' etc. It's unclear which directory would be equivalent to the "libraries" folder.

[image]

sonofevil, 2025-03-10:
Ofc You can always just change the variables to wherever you want:

`require_once "/usr/lib/php/modules/adb.php";`

`$adb = new ADB("$HOME/libraries/");`

demon3z, 2025-04-07:
for windows,new version php ,such as 8.4.5 .you should have the "libraries" dir.and put the platform-tools file in it.
对于windows系统，应该新建文件夹libraries，并将platform-tools中的文件放入其中

[image]