MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #84: undifined function openssl_decrypt() ISSUE

resakastepme, 2024-03-20:
[image]

i have already check the php.ini file for enabling the openssl and curl extension as what the readme says..
any solution to this issue? have anyone else face the same problem as me?

cikgit, 2024-03-21:
> ![image](https://private-user-images.githubusercontent.com/88456993/314766288-0f208cb8-1b9c-42ff-b5d9-8d3c0033c399.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

go to this post and read the instructions
https://xiaomi.eu/community/threads/free-official-hyperos-unlock-bootloader.71323/page-5#post-720783

i think settings.apk must be modified according to this post
https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/82

Keronaya, 2024-03-22:
> > ![image](https://private-user-images.githubusercontent.com/88456993/314766288-0f208cb8-1b9c-42ff-b5d9-8d3c0033c399.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwMzQ1NTIsIm5iZiI6MTcxMTAzNDI1MiwicGF0aCI6Ii84ODQ1Njk5My8zMTQ3NjYyODgtMGYyMDhjYjgtMWI5Yy00MmZmLWI1ZDktOGQzYzAwMzNjMzk5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDE1MTczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5ZTFlNzBlZDlkYjMyODEzZDkxMWVjNzViNzE3ZDBhZTQyZDA2OGUxZTMwZTk1NDIzOGVjZGNhYmYxM2RiZTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.T373x5BDOIAoomcyZzBuM1mVoao-kXNNBbkWc0v_BYA)
> > i have already check the php.ini file for enabling the openssl and curl extension as what the readme says.. any solution to this issue? have anyone else face the same problem as me?
> 
> go to this post and read the instructions https://xiaomi.eu/community/threads/free-official-hyperos-unlock-bootloader.71323/page-5#post-720783
> 
> i think settings.apk must be modified according to this post #82 but downloading files from chinese services is a pain in the *ss

I think his issue still about the php problem or network problem, not the 1000 error. 
**For PHP problem you should copy your PHP**  files 'https://www.php.net/downloads' to C and rename like this 'C:\php\ext\curl'
[image]

For network problem you should try to change another smi-card.
.

xbn2002, 2024-05-05:
Look #91 pls

closed 2024-07-02 by MlgmXyysd