MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #85: Fail to send request, check your internet connection

THX-UltraII, 2024-03-22:
Hi,

I get this when trying to bypass:

[image]

THX-UltraII, 2024-03-22:
So, I managed to fix the "check your internet connection" notification. Of course it would be way too easy if it worked now :D:D

Next challenge: "Request parameter error (10000)"

Anyone tips?
[image]

zaaloul, 2024-03-24:
> Hi,
> 
> I get this when trying to bypass:
> 
> ![Schermafbeelding 2024-03-21 145324](https://private-user-images.githubusercontent.com/164478224/315934812-c031e6a8-0e67-4df2-b598-e59b8cf696c6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEyNjMxMjksIm5iZiI6MTcxMTI2MjgyOSwicGF0aCI6Ii8xNjQ0NzgyMjQvMzE1OTM0ODEyLWMwMzFlNmE4LTBlNjctNGRmMi1iNTk4LWU1OWI4Y2Y2OTZjNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNFQwNjQ3MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZWYzNGZhNDM5NWY2NGNjM2YyNjBkNTIzZTI2YmEwMDZlOWM1OTIyNWIwNjA5NWQwMTEwMzc1ODgwN2E3MjFmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.c_u53BB1gcu3VwJzzF3dJHcccE3cYH_IlXHH2ZFn33U)

How did You manage to get over it man ? I am facing the same network failure

Tigercat000, 2024-04-22:
How to fix "Fail to send request, check your internet connection"

xbn2002, 2024-05-05:
> How to fix "Fail to send request, check your internet connection"

Make sure that your internet is stable and quick
Another possible way is to change the bind server to international, by modifying the option in "bypass.php"

xbn2002, 2024-05-05:
> So, I managed to fix the "check your internet connection" notification. Of course it would be way too easy if it worked now :D:D
> 
> Next challenge: "Request parameter error (10000)"
> 
> Anyone tips? ![Schermafbeelding 2024-03-22 184319](https://private-user-images.githubusercontent.com/164478224/316121353-84478e05-8ae5-4f04-9e30-80175fa02a36.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ4OTc4MjgsIm5iZiI6MTcxNDg5NzUyOCwicGF0aCI6Ii8xNjQ0NzgyMjQvMzE2MTIxMzUzLTg0NDc4ZTA1LThhZTUtNGYwNC05ZTMwLTgwMTc1ZmEwMmEzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNVQwODI1MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MzMxNGI5YmIwMDdhZmNjMjI2ODdjMzAwOTJhM2E3YmVkOTdlMjcyYWVmYWFkYmUwNzA4MzA1ZWZhNTlhOWY4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-8iJpQsAFqf6Uv5XxlOdEyqgOYbFVTlEncpn80bTGGg)

look #82 pls

LeafShinobi96, 2024-09-24:
> > How to fix "Fail to send request, check your internet connection"
> 
> Make sure that your internet is stable and quick Another possible way is to change the bind server to the international one, by modifying the option in "bypass.php"

How do I do this? I constantly get this error on multiple Mi accounts.

Baglanoc, 2024-10-07:
[image]
Please help 😔

zyrta, 2024-10-25:
[image]
help

Puneetdp, 2024-12-03:
> > How to fix "Fail to send request, check your internet connection"
> 
> Make sure that your internet is stable and quick
> Another possible way is to change the bind server to the international one, by modifying the option in "bypass.php"

What we need to edit in "bypass.php". ??

carlos24ramirez, 2025-01-19:
[image]
[image]

carlos24ramirez, 2025-01-19:
Cuál sería la solución a esto ?

carlos24ramirez, 2025-01-19:
[image]

ahmedhanbal, 2025-05-23:
Open and edit bypass.php; set $useGlobal = true and then run bypass.cmd again. it will be fixed.

[image]

[image]