MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #86: wating for device connection 怎么办，很急，谢谢大佬

Ezgx, 2024-03-24:
[image]

LeanFly, 2024-03-27:
usb连接问题？设备没连上

SabinLee, 2024-03-28:
应该是驱动有问题识别不到？可以尝试重装一下驱动

FrzMtrsprt, 2024-04-13:
可能是adb没放对位置，应该把platform-tools里的adb.exe等文件解压到libraries目录，而不是把platforms-tools文件夹放到libraries目录下。

xbn2002, 2024-05-05:
platform-tools 没有配置好。另外你可以去论坛上搜其他人整合好的版本，可以直接用，不需要自己配环境。

jeromelgh, 2024-10-10:
> Maybe adb is not placed in the right place. You should unzip the adb.exe and other files in the platform-tools to the libraries directory instead of putting the platforms-tools folder in the libraries directory.

This got me past the device connection error. I am now stuck on "Request parameter error (10000)"

FrzMtrsprt, 2024-10-10:
> This got me past the device connection error. I am now stuck on "Request parameter error (10000)"

https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/tag/1.0-fix

tcblaize, 2024-10-10:
> > This got me past the device connection error. I am now stuck on "Request parameter error (10000)"
> 
> https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/tag/1.0-fix

Does this settings file work for a Redmi Note 13 Pro+ with android 14

FrzMtrsprt, 2024-10-10:
> Does this settings file work for a Redmi Note 13 Pro+ with android 14

I don't have a Redmi Note 13 Pro+ with android 14 so idk.

tcblaize, 2024-10-10:
> > Does this settings file work for a Redmi Note 13 Pro+ with android 14
> 
> I don't have a Redmi Note 13 Pro+ with android 14 so idk.

I'm asking if it's a universal file, or there;s a different one for each phone

FrzMtrsprt, 2024-10-10:
> I'm asking if it's a universal file, or there;s a different one for each phone

It's not, it may or may not work for you depending on your os version and your model.

cute-omega, 2024-12-29:
#236