MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #88: Request parameter error (10000)

Sreehari1710, 2024-03-25:
I tried the settings.apk in the solved section but after installing the apk i wasn't able to access the developer settings.

[2024-03-25] [10:28:00] [INFO] - Account bind request found! Let's block it.
[2024-03-25] [10:28:00] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333
[2024-03-25] [10:28:00] [INFO] - Sending POST request...
[2024-03-25] [10:28:00] [INFO] - Request parameter error (10000)

Ryancheng236, 2024-03-25:
I got same problem. Do you resolve this?

LeanFly, 2024-03-27:
https://www.123pan.com/s/625SVv-BYxbh.html

从上面网盘下载setting.apk，安装后重启手机

Ryancheng236, 2024-03-27:
谢谢啦！！！

 
https://www.123pan.com/s/625SVv-BYxbh.html

ZawPhoneMaw, 2024-03-27:
I got same issues. After installing setting.apk, can't open developer option. Crashing again & again
Any solution for unlock bootloader Plz!!!

ketor, 2024-03-31:
> I got same issues. After installing setting.apk, can't open developer option. Crashing again & again Any solution for unlock bootloader Plz!!!

Got same issues on Redmi K60 Ultra( Running HyperOS 1.0.9.0)。 Maybe this solution is not work now.

MoutainPL, 2024-04-08:
> https://www.123pan.com/s/625SVv-BYxbh.html
> 
> 从上面网盘下载setting.apk，安装后重启手机

not work

LananaHWP, 2024-04-15:
> > https://www.123pan.com/s/625SVv-BYxbh.html
> > 从上面网盘下载setting.apk，安装后重启手机
> 
> not work

disable bluetooth, then it works

ketor, 2024-04-20:
After download settings.apk from the 123pan,  I have unlocked K60 ultra with the latest hyperos 1.0.11.0.
Thanks very much!

closed 2024-07-02 by MlgmXyysd

eddiemukri01, 2024-07-17:
[image]

How can i install setting.apk?

bdd786, 2025-01-29:
How can we install settings.apk, it says app not installed as package appears to be invalid. It tries to update the original settings app