MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #91: Uncaught Error: Call to undefined function openssl_decrypt()

alva1018, 2024-04-05:
[image]
出现这个问题，php.ini设置里curl和openssl都打开了
[image]

请教解决方法，多谢多谢！！

alva1018, 2024-04-05:
已解决。。配置文件路径问题，放入C盘根目录就OK了，惭愧

W1LLY-W0N, 2024-04-28:
Hello !

I think I am experiencing the same issue as you. How did you do to solve it ?

alva1018, 2024-04-29:
Put all the files into the directory "C:" should resolve the problem. It seems to be related to the address specified in the configuration file.

TriarOriginalSin, 2024-06-30:
我也遇到了相似的问题，把文件夹或文件放入C:\均无效，求助

closed 2024-07-02 by MlgmXyysd

Abdurrahman98XX, 2025-01-09:
same problem here, why is it close if there is not an answer?