MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #92: 小米13等早期为MIUI出厂，后期为HyperOS出厂的设备可以解锁吗？

Patrick-Ze, 2024-04-06:
感谢大佬开发的这个项目，近期想换手机，解锁对我来说是刚需，希望在下单前调查好解锁能力。

Google了一下相关问题但是没有找到特别准确的回答，也找了相关的issue，看到下面这个说明

> 出厂MIUI的老设备堵不了，除非把MIUI的解锁限制提升到和HyperOS一样。出厂就是HyperOS自然不会存在MIUI的版本，所以可以把这些设备拉个清单，强制走校验逻辑

https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/18#issuecomment-1864863329

有两点疑惑请教一下：

1. `30001`强制验证是基于出厂验证还是基于设备当前系统呀？假设一台出厂为MIUI的手机在未解锁状态下升级到了HyperOS，会受新的解锁规则限制么？
2. 似乎后期生产的小米13都是出厂HyperOS，那这批手机也是解锁不了的是么？（本以为小米13因为有MIUI出厂的版本，还有解锁可能）

chengnan049, 2024-04-20:
1.是，规则写了啊
2.理论是的

ketor, 2024-04-20:
简单来说是按照机器的型号来确定的，如果这个型号发布的时候系统就是HyperOS，就是无法bypass的，如果发布时是MIUI，即使后期新发货的机器预装了HyperOS也还是可以bypass的。

看主流型号的话，K70系列，Mi14系列都无法bypass。K60系列，Mi13系列以及更早的记性都可以bypass。解锁还是要趁早，现在新版的HyperOS需要自己降级一下Settings.apk才可以绑定账户了，如果未来mi家改了Settings的版本号，导致无法降级了，那就可能无法bypass了。

Patrick-Ze, 2024-05-01:
@ketor 感谢您的解答~

closed 2024-05-01 by Patrick-Ze