MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #93: Stuck on "you can bind account..."

YaroslavQw, 2024-04-07:
Stuck on "you can bind account in the developer options"
Device Xiaomi pad 6 pro
Rom hyperos 1.0.4.0cn

YaroslavQw, 2024-04-07:
[image]

kbiits, 2024-04-07:
It's waiting you to try to add your mi account. You need to interact with your phone, go to mi unlock status and try to add account.

YaroslavQw, 2024-04-07:
Unfortunately, the problem is that nothing happens after I try to add an account in the developer settings "couldn't add. Please go to mi community".

kbiits, 2024-04-07:
Yes, that is the purpose. We need to intercept the request when you try to add an account, make a modification to the request, and send it again. So, it's expected to fail, just see your terminal output

kbiits, 2024-04-07:
If it succeeded you will see message "Your account was bound successfully" in your terminal output

YaroslavQw, 2024-04-08:
But the problem is that nothing is happening. Nothing at all

YaroslavQw, 2024-04-08:
It's fix after format data

closed 2024-04-08 by YaroslavQw

yash96644, 2026-04-05:
did u figure out this issue