MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #95: Request parameter error (10000)

Cromedex, 2024-04-09:
I am currently trying to unluck the bootloader on this mf xiaomi (my dad gave it to me but fuck the ads). I went with all the procedure and preparatives but now I am stuck with the bootloader remaining locked, I went with this bypass but now there is this error:

[2024-04-09] [20:14:02] [INFO] * Now you can bind account in the developer options.
[2024-04-09] [20:14:17] [INFO] - Account bind request found! Let's block it.
[2024-04-09] [20:14:17] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on value of type null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on value of type null in C:\php\bypass.php on line 333
[2024-04-09] [20:14:17] [INFO] - Sending POST request...
[2024-04-09] [20:14:17] [INFO] - Request parameter error (10000)

now what?

smile-YYY, 2024-04-22:
You can install the new settings app, then log in to your Xiaomi account again, connect the phone to the computer to unlock it, this is how I solved the problem

p-hao, 2024-04-27:
> You can install the new settings app, then log in to your Xiaomi account again, connect the phone to the computer to unlock it, this is how I solved the problem

Where can i get the `settings app`

p-hao, 2024-04-27:
> > You can install the new settings app, then log in to your Xiaomi account again, connect the phone to the computer to unlock it, this is how I solved the problem
> 
> Where can i get the `settings app`

Ok, found the result in issue #82

closed 2024-07-02 by MlgmXyysd

carlos24ramirez, 2025-01-19:
Tengo el error 10000 en hyperos 2 pero el setting.apk no deja instalarlo en rom global para poco f6 pro