MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #96: 点击绑定账号和设备后数据无法解析

themismin, 2024-04-14:
mac系统， 红米13 pro

```adb devices -l```
List of devices attached
21328b76               device usb:336592896X product:garnet model:2312CRAD3C device:garnet transport_id:4

> 优化了部分代码后解析出来 devices
```
Array
(
    [0] => 21328b76
    [1] => device
    [2] => usb:336592896X
    [3] => product:garnet
    [4] => model:2312CRAD3C
    [5] => device:garnet
    [6] => transport_id:4
)

Array
(
    [0] => Array
        (
            [serial] => 21328b76
            [status] => device
            [transport] => 4
            [manufacturer] => Xiaomi
            [brand] => Redmi
            [board] => garnet
            [name] => garnet
            [product] => garnet
            [model] => 2312CRAD3C
            [device] => garnet
        )

)
```

> 点击绑定账号和设备后

> args 数据

$args = '#&^GEkTXewsJE5z+Z+j0Pvax3t7/WTE/fSYC0zI5EJmswZxFgjHHpOjq5T1drjeCnY0GE/whNo/uClX1b7y61ZndWZETRtF//4kqrIwhG0hGbBFfJ0uGLWOSW302NcMq/EO8lVzJjhCi21sN+OKaYV+8HW2dAvweHD0/xGOY7HCi5hTTgKan3WVZK7zfcEm5NaTi15mrQpf3sjJh9Otdm2r2+F7B9w7ZBa/VxpQkYzweuInmVLPKmubTpFrzrFoAKDXfdRenxePo75PtmxyvJSaJifBN4hUzQWO/dOiJp8C+XzocoqOPrEwd56CMnRaIaAJ4J30PKVrGTpnz1LiS4C/rA==!!IDukdCEplzYTpH4m0vHyGC1j1mWZSUvTvQ4+lbbciuw3R9Kb4rIiWj0HCwnnE3L3/yS8Ks3/2fW+uT0dS1uIVzrZ0Wkumsf+yxW7z4h48jxzyizQ31/aT/oFJCFu4WKiMOIO1kGqA2r2PFSzLaTkXhel+tsSfRDOf5dbDzjYbXXcDFHdDfxWmkW4YXA7JoOIeF3djue3vO1S70eR2CkVP9Y5Bn135roxFT8ReaKpg4fHV0Imu87Keax0PESyE4ECLEhN/M7to4a6jFOl0kf0SnEBqb935KS5OEEI5V+rwntGnGqaJtUmhGiXLq71/10eULM5pal7vObO4WqYCeHC7pkOh+Yoz5m2tH1i5FIEptpWBKd7q+IC3VKz0rcjlmcQuUk7hzFiqY0VM75f29WVThnas8/xhUM2/d2efoUoEaKXrWUhsSzxKjzK7gElmtF76Byy6tY+5RzTG2t9qhBtd7m+HGue9JJ1Xq3+Q8rpgBJOuXWOhznGWkLquBNZKEgy7yGA0xec5+lg+ZTQjsPpyc9h8xXasH9dRY67G0w4MXwNKZ8MJ7ppqkmAKhLycxMyVdlT+gbT4C5orBF7FsdMCmYpMYFicEptqMVLGLWsYWJChdf/iSn6f0Js/xzu2BbgNZ2tYQOAQRIRu98Qtcn+ENsz3C4u0Gzu/0lYELENIOfNlopw6NHIsn9fbXvQm8GmZ48/puKYy+vC4mpaWIn8gadbaSQBYQMB/WDMmbfWky2gurww8clfuDnHw2ePTxoWB1RtdDhB1N8fDEpQPPRjLzjlcgt8zLVJIsE4gSrlfr2GpbnqDU35Zt7qTwfMY15o8ukroi41IM3Cz1sqKdCLHRofbfGS4R0ZzoohhjsKWO2Tj6u2LDTaw6RapAgSGeVIMniH2ykqKUPacpjHGk/HKQ==^&#';

decryptData($args) 返回 bool(false)

> headers 数据

$headers = '#&^J/SH7YSjC/ONxo7A5qe8Vq33HkkkBcz5NH4DO08/5XqJ6k4VXxh72aIi8bI1fh0FYW5ywijyWejo9fGQqJEhp6zyWXsz3q9FDDgZeZd/k2+zAcGLq0QzcckY6S04G/7VsMLK3ZWMJJBUJrgr8A5Z0rsASLQXVoq5yhgGJqya6hap8+uEiT39XdJM7LNsGnYRQD3vmEaoIKdxrDYmeCMLqkIrJ0R14K4mMFsMkzYx1MzGmWYrlrD32FraTjEqsZM9idzi+TmHH0AvMAsv4p4G/C/FUkaU7Vrt49jSgR+gbie00POHsuHb2A6rFPec1yIkkqmAl4VtfeD+OrS9ZuZaqg==!!EblNrC6TyYJAFxpQ7qsRbV8jw/pKsOmrIomek2U16hx6ufrXzdsAOSZ42RAe6wRNId1pcZK2RmPCEkSi+JS5p+KCXu2ZB7Yb0MAjZRn+wQXTV1LZM8hh1rhY1JXgWkZmYJx+useOpjX46C8t7muBSFEiOoY6v/iPDHE+3GBKe99fhHtt1wVorfy/L6Kb9KbWGU9u9LANunHmFnloPOYirFEFO1VHin3U/vFm9ArDwgzN7gTxLV+dHnz/UAecQzDF1RiaSgZnbFw3m8Mr93t3cVvoIdzdWDbp0SD+Mm0ohnsX6s0h9DYwe05GOJDdvkuN2a2pgiysRqx35oLr0xn+lsKS6A58rj5S+t/F1aWkjmBpGtrDygdxkGXY59hOGL5Ve2I8GIp6+pOqsio0WMHZj6YKpRohb4CX+h5aBvuHtuC2xbRjBHcEdm0iN2rtE3QDwvm5br3h2WKacOEM6+493OOx5ZmNhCrzoQgVtOHSKNasvT7mRyPbEIhHTsTGscXq^&#';

decryptData($headers); 返回 bool(false)

themismin, 2024-04-14:
[Xiaomi-HyperOS-BootLoader-Bypass.zip](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/files/14970557/Xiaomi-HyperOS-BootLoader-Bypass.zip)

我调试的代码

themismin, 2024-04-14:
### 系统版本
Xiaomi HyperOS
1.0.1.0.UNRCNXM

xinshengsiyu, 2024-04-16:
修改bypass.php   279行的代码： true => false
$id = $adb -> getDeviceId($device[1], false);

安装新的设置apk
https://www.123pan.com/s/625SVv-BYxbh.html

然后重新操作就行了。

sword-jin, 2024-04-17:
同样的问题，修改 279 行后，卡在

```
[2024-04-17] [02:05:44] [INFO] - Processing device UC8DYHW49LI7AMNN(4)...
```

@xinshengsiyu

sword-jin, 2024-04-17:
@themismin 请问你修好了吗

xinshengsiyu, 2024-04-17:
@sword-jin 你安装了新的设置apk没，可以贴下完整的日志。

sword-jin, 2024-04-17:
@xinshengsiyu 
感谢回复，说实话，我没有明白安装新的 设置apk 是什么意思。设置 apk 是什么 app 呢

sword-jin, 2024-04-17:
成功了，得到了最终不行的 30001 错误。fuck xiaomi

themismin, 2024-04-17:
[2024-04-17] [16:50:15] [INFO] - Sending POST request...
[2024-04-17] [16:50:16] [INFO] - Request parameter error (10000)
验证失败，重启，系统恢复出厂设置都不行，还是报 10000
@xinshengsiyu @sword-jin

xinshengsiyu, 2024-04-18:
@themismin  安装新的设置apk没。 安装之后再重新登录小米账号。 然后执行脚本，我是这么操作的就可以了

hanxiang-li, 2024-04-19:
30001 真没解决方案吗

gsmlm, 2024-04-22:
$args = '#&^ 这种的都用新算法了。没戏了！

之前是这样的
public class AESUtil	// class@000600 from classes2.dex
{

    public static String encrypt(String p0,String p1){
       if (p1 == null) {
          throw new Exception("AES ENCRYPT : sKey is null");
       }
       if (p1.length() != 16) {
          throw new Exception("AES ENCRYPT : sKey\'s length is not 16");
       }
       Cipher instance = Cipher.getInstance("AES/CBC/PKCS5Padding");
       instance.init(1, new SecretKeySpec(p1.getBytes(), "AES"), new IvParameterSpec("0102030405060708".getBytes()));
       byte[] uobyteArray = instance.doFinal(p0.getBytes());
       return Base64.encodeToString(uobyteArray, 2);
    }
    public static String getDefaultAESKeyPlaintext(){
       return "20nr1aobv2xi8ax4";
    }
}

新加密的是这样的：
class LogEncryptor	// class@001541 from classes2.dex
{
    private final String mEncrytedKey;
    private final SecretKey mSecretKey;
    private static final byte[] SYM_ENCRYPT_ALGORITHM_IV;

    static {
       LogEncryptor.SYM_ENCRYPT_ALGORITHM_IV = "bootloaderXiaomi".getBytes();
    }
    public void LogEncryptor(){
       super();
       byte[] uobyteArray = Base64.decode("MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAxPEmV1vZ60qc39gWvaSc\n7QgV/Ltc95eTBiWsRcN5VDeqjGwRPmk7TBXvU+YQ6q2LrfiaDQYg8ZwxjwUTsWoL\nJ7l8AHE0WdUEvdV36+BMbB9w7ts2IISZZNnJyyZleU+SImWYRybKkTPX//Ld/bgK\nNFz3dxJzYxLXdKzcZogHLI2Mvvj31/ZmqvKuRxXBQ2iU4oSPthQRXFY+KbQJ1Z3Z\nsFzMJfGaY1jj+8ymUd4zWGXgztQLuvpUNtiVHGW1WhP8854yJqbQ1VcqfIueKR74\nqoQgUbXHFuYbvz6B0c+bEgJ/tn/bXcM8Zo8aADFgZNCChbzAhB9wf3zx2RLJe7aN\nawIDAQAB", 0);
       try{
          KeyFactory instance = KeyFactory.getInstance("RSA");
          try{
             PublicKey publicKey = instance.generatePublic(new X509EncodedKeySpec(uobyteArray));
             try{
                KeyGenerator instance1 = KeyGenerator.getInstance("AES");
                instance1.init(256);
                SecretKey secretKey = instance1.generateKey();
                try{
                   this.mSecretKey = secretKey;
                   Cipher instance2 = Cipher.getInstance("RSA/ECB/PKCS1Padding");
                   try{
                      instance2.init(1, publicKey);
                      this.mEncrytedKey = Base64.encodeToString(instance2.doFinal(secretKey.getEncoded()), 2);
                      return;
                   }catch(javax.crypto.IllegalBlockSizeException e4){
                      throw new RuntimeException("Should never happen. ", e4);
                   }catch(javax.crypto.BadPaddingException e0){
                   }
                }catch(javax.crypto.NoSuchPaddingException e4){
                   throw new RuntimeException(e4);
                }catch(java.security.NoSuchAlgorithmException e0){
                }catch(java.security.InvalidKeyException e4){
                   throw new IllegalArgumentException("Encrypt log RSA public key is not valid. ", e4);
                }
             }catch(java.security.NoSuchAlgorithmException e4){
                throw new RuntimeException(e4);
             }
          }catch(java.security.spec.InvalidKeySpecException e4){
             throw new IllegalArgumentException("The public key not valid. ", e4);
          }
       }catch(java.security.NoSuchAlgorithmException e4){
          throw new RuntimeException(e4);
       }
    }
    private String encryptMsg(String p0){
       String str = "Should never happen. ";
       try{
          Cipher instance = Cipher.getInstance("AES/CBC/PKCS5Padding");
          try{
             LogEncryptor tmSecretKey = this.mSecretKey;
             IvParameterSpec ivParameterS = new IvParameterSpec(LogEncryptor.SYM_ENCRYPT_ALGORITHM_IV);
             try{
                instance.init(1, tmSecretKey, ivParameterS);
                tmSecretKey = StandardCharsets.UTF_8;
                tmSecretKey = p0.getBytes(tmSecretKey);
                tmSecretKey = instance.doFinal(this);
                tmSecretKey = Base64.encodeToString(this, 2);
                return this;
             }catch(javax.crypto.BadPaddingException e4){
                throw new RuntimeException(str, e4);
             }catch(javax.crypto.IllegalBlockSizeException e0){
             }
          }catch(java.security.InvalidAlgorithmParameterException e4){
             throw new RuntimeException(str, e4);
          }catch(java.security.InvalidKeyException e0){
          }
       }catch(java.security.NoSuchAlgorithmException e4){
          throw new RuntimeException(str, e4);
       }catch(javax.crypto.NoSuchPaddingException e0){
       }
    }
    public String wrapEncryptMsg(String p0,String p1){
       LogEncryptor logEncryptor;
       try{
          Object[] objArray = new Object[4];
          objArray[0] = "#&^";
          objArray[1] = this.mEncrytedKey;
          logEncryptor = this.encryptMsg(p1);
          objArray[2] = this;
          logEncryptor = "^&#";
          objArray[3] = logEncryptor;
          logEncryptor = String.format("%s%s!!%s%s", objArray);
          return this;
       }catch(java.lang.Exception e4){
          logEncryptor = new LogEncryptor{e4};
          logEncryptor = String.format("Failed to encrypt the message: %s. ", this);
          Log.e(p0, this);
          logEncryptor = "Log record failure";
          return logEncryptor;
       }
    }
}

secretKey 通过RSA加密得到mEncrytedKey放在数据头部，后面放加密后的数据，服务器通过私钥解密头部的mEncrytedKey，用来解密后面的数据。没有私钥没办法替换了！

sword-jin, 2024-04-23:
@gsmlm 没有办法拿到本地的私钥吗

xinshengsiyu, 2024-04-23:
感觉可以吧，我刚用我的小米13成功了。
可以加个飞书交流下 
[image]

gsmlm, 2024-04-23:
> @gsmlm 没有办法拿到本地的私钥吗

私钥肯定在服务器，既然用rsa算法了，肯定不会把私钥放本地的。现在唯一的办法就是替换早一点版本的Settings.apk，不知道还有没有大神会别的方法。动态调试断点取secretKey或者hook。但是不root好像不容易hook吧。大神很多，我等菜鸟只能等大神们造轮子了。

hanxiang-li, 2024-04-23:
> > @gsmlm 没有办法拿到本地的私钥吗
> 
> 私钥肯定在服务器，既然用rsa算法了，肯定不会把私钥放本地的。现在唯一的办法就是替换早一点版本的Settings.apk，不知道还有没有大神会别的方法。动态调试断点取secretKey或者hook。但是不root好像不容易hook吧。大神很多，我等菜鸟只能等大神们造轮子了。

小米14Pro 有没有早一点的settings

gsmlm, 2024-04-23:
@hanxiang-li 
> 小米14Pro 有没有早一点的settings
没有啊，这是别人整理的：https://www.123pan.com/s/625SVv-BYxbh.html
https://pan.quark.cn/s/16bbc6281511#/list/share

hanxiang-li, 2024-04-23:
> @hanxiang-li
> 
> > 小米14Pro 有没有早一点的settings
> > 没有啊，这是别人整理的：https://www.123pan.com/s/625SVv-BYxbh.html
> > https://pan.quark.cn/s/16bbc6281511#/list/share

gg,不搞了

gsmlm, 2024-04-23:
> > @hanxiang-li
> gg,不搞了
哈哈可以自己下载早起版本的刷机包提取看看呀！我也懒得搞了 今早8点刚答题考试完，不知道能通过不！

hanxiang-li, 2024-04-23:
> > > @hanxiang-li
> > > gg,不搞了
> > > 哈哈可以自己下载早起版本的刷机包提取看看呀！我也懒得搞了 今早8点刚答题考试完，不知道能通过不！

md,他那个八点就是故意恶心人了，都不想去答题了，老是错过时间

gsmlm, 2024-04-23:
> > > > @hanxiang-li
> md，他那八点就是做饭了，不想回答问题了，老是错过时间
确实，定了几个闹铃，结果查两道填空题还没填完。那题目和刷机开bl都没关系了！离谱

Tigercat000, 2024-04-23:
> @hanxiang-li
> 
> > 小米14Pro 有没有早一点的settings
> > 没有啊，这是别人整理的：https://www.123pan.com/s/625SVv-BYxbh.html
> > https://pan.quark.cn/s/16bbc6281511#/list/share

誰能找到K70 pro 的setting.apk 嗎?

gsmlm, 2024-04-25:
@hanxiang-li @Tigercat000 
You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck)
这个我测试通过了

hanxiang-li, 2024-04-25:
> @hanxiang-li @Tigercat000 You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck) 这个我测试通过了

请问你的是小米14Pro吗？能给个下载地址吗？

gsmlm, 2024-04-25:
@hanxiang-li
> 请问你小米是14Pro吗？给个下载地址吗？
https://github.com/TheAirBlow/HyperFuck
这里面都有，Settings.apk 通用，我的mix fold2 测试通过！

Tigercat000, 2024-04-25:
> @hanxiang-li
> 
> > 请问你小米是14Pro吗？给个下载地址吗？
> > https://github.com/TheAirBlow/HyperFuck
> > 这里面都有，Settings.apk 通用，我的mix fold2 测试通过！

就用他內的setting.apk? 我下載了不能安裝?

hanxiang-li, 2024-04-25:
> @hanxiang-li
> 
> > 请问你小米是14Pro吗？给个下载地址吗？
> > https://github.com/TheAirBlow/HyperFuck
> > 这里面都有，Settings.apk 通用，我的mix fold2 测试通过！

[image]
我失败了，30001

Tigercat000, 2024-04-25:
> ![image](https://private-user-images.githubusercontent.com/45918130/325465476-6da8cc2f-3dfa-442e-8845-42cc1e042f6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

用這方法，k70 pro 也是30001

gsmlm, 2024-04-25:
@Tigercat000 
> 就用他內的setting.apk? 我下載了不能安裝?
你看下 是100多m的Settings.apk吗？如果不是点击文件 单独下载

gsmlm, 2024-04-25:
@hanxiang-li @Tigercat000 
这个错误不知道咋办了，是不是社区等级没到5级的校验了？我的小米社区到5级了

Tigercat000, 2024-04-25:
> @Tigercat000
> 
> > 就用他內的setting.apk? 我下載了不能安裝?
> > 你看下 是100多m的Settings.apk吗？如果不是点击文件 单独下载

是134B 但都不能安裝?如用他方法，會出現這:
PS C:\F> pip install pure-python-adb PyCryptoDome
pip : The term 'pip' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the s
pelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ pip install pure-python-adb PyCryptoDome
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pip:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Tigercat000, 2024-04-25:
> @hanxiang-li @Tigercat000 这个错误不知道咋办了，是不是社区等级没到5级的校验了？我的小米社区到5级了

那能幫我解K70 Pro 嗎?

gsmlm, 2024-04-25:
@Tigercat000
> 那能幫我解K70 Pro 嗎?
没办法呀！只能把自己账号升级到5级试试了。现在签到+浏览帖子+看指定视频5分钟 都可以获得成长值。

Tigercat000, 2024-04-25:
> @Tigercat000
> 
> > 那能幫我解K70 Pro 嗎?
> > 没办法呀！只能把自己账号升级到5级试试了。现在签到+浏览帖子+看指定视频5分钟 都可以获得成长值。

都明白，謝謝

Tigercat000, 2024-04-25:
[2024-04-25] [05:06:42] [INFO] - Processing device 6772ab1d(1)...
[2024-04-25] [05:06:42] [INFO] - Finding BootLoader unlock bind request...
[2024-04-25] [05:06:43] [INFO] * Now you can bind account in the developer options.
[2024-04-25] [05:06:47] [INFO] - Account bind request found! Let's block it.
[2024-04-25] [05:06:47] [INFO] - Refactoring parameters...
[2024-04-25] [05:06:47] [INFO] - Sending POST request...
[2024-04-25] [05:06:50] [INFO] - Binding failed, this device has been forced to verify the account qualification by Xiaomi. (30001)

hanxiang-li, 2024-04-25:
> @hanxiang-li @Tigercat000 这个错误不知道咋办了，是不是社区等级没到5级的校验了？我的小米社区到5级了

还得答题通过才可以，这政策就是个狗屎，已经违背了最初的初衷

Tigercat000, 2024-04-25:
> > @hanxiang-li @Tigercat000 这个错误不知道咋办了，是不是社区等级没到5级的校验了？我的小米社区到5级了
> 
> 还得答题通过才可以，这政策就是个狗屎，已经违背了最初的初衷

對，是把小米最強的限制了，會逐漸把米粉趕到其他品牌門下

gsmlm, 2024-04-25:
> > 还得答题通过才可以，这政策就是个狗屎，已经违背了最初的初衷
> 
> 對，是把小米最強的限制了，會逐漸把米粉趕到其他品牌門下
哈哈，别的还有什么牌子开放bl门槛低的？

hanxiang-li, 2024-04-25:
> > > 还得答题通过才可以，这政策就是个狗屎，已经违背了最初的初衷
> > 
> > 
> > 對，是把小米最強的限制了，會逐漸把米粉趕到其他品牌門下
> > 哈哈，别的还有什么牌子开放bl门槛低的？

貌似一加还可以直接解锁

themismin, 2024-04-27:
> [xinshengsiyu](/xinshengsiyu)

30001 了，能解决吗

xbn2002, 2024-05-05:
> 30001 真没解决方案吗

没有。以后也不会有

xbn2002, 2024-05-05:
> > [xinshengsiyu](/xinshengsiyu)
> 
> 30001 了，能解决吗

不能，没办法解决，出厂澎湃的不行

xbn2002, 2024-05-05:
> 用這方法，k70 pro 也是30001

出厂澎湃的不行，已经说过很多遍了呀

xbn2002, 2024-05-05:
> ![image](https://private-user-images.githubusercontent.com/45918130/325465476-6da8cc2f-3dfa-442e-8845-42cc1e042f6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

出厂澎湃不行

cnfatal, 2024-06-28:
有最新的settings.apk 吗，我研究下

5918vo, 2024-07-06:
xiaomi 13 Request parameter error (10000) 
可以使用下载安装setting.apk方式解决 (能解密成功)

Xiaomo3250, 2024-07-08:
@xinshengsiyu 佬，对不起打扰了，但是还是想问一下这个Settings.apk安装之后点击“开发者模式”，设置就会闪退是什么原因呢？

我的手机型号是 Redmi K50 ，下载的“红米K50至尊_K50G_note12turbo”的设置文件，使用 ADB install 安装，现在的 HyperOS 版本是 1.0.6.0.ULNCNXM。

闪退之后的解决方案是从应用设置里卸载更新，暂时还没有恢复出厂设置。

（对不起因为还有一点数据没有转移所以还没有来得及恢复出厂设置，我后续再看看这个问题，谢谢佬的工具包）

Xiaomo3250, 2024-07-08:
> @xinshengsiyu 佬，对不起打扰了，但是还是想问一下这个Settings.apk安装之后点击“开发者模式”，设置就会闪退是什么原因呢？
> 
> 我的手机型号是 Redmi K50 ，下载的“红米K50至尊_K50G_note12turbo”的设置文件，使用 ADB install 安装，现在的 HyperOS 版本是 1.0.6.0.ULNCNXM。
> 
> 闪退之后的解决方案是从应用设置里卸载更新，暂时还没有恢复出厂设置。
> 
> （对不起因为还有一点数据没有转移所以还没有来得及恢复出厂设置，我后续再看看这个问题，谢谢佬的工具包）

@xinshengsiyu 打扰了，我已经根据adb的调试信息定位了问题出在旧版本设置中蓝牙与当前设置不同导致的崩溃，进而造成开发者模式中无线调试模块崩溃，关闭蓝牙后就可以正常使用了

打扰致歉

7HE-W0R1D, 2024-07-11:
> 修改bypass.php 279行的代码： true => false $id = $adb -> getDeviceId($device[1], false);
> 
> 安装新的设置apk，安装后重新登录下 https://www.123pan.com/s/625SVv-BYxbh.html
> 
> 然后重新操作就行了。

谢谢 2024.6.11 小米Pad6绑定成功

VwEl, 2024-07-16:
> 修改bypass.php 279行的代码： true => false $id = $adb -> getDeviceId($device[1], false);
> 
> 安装新的设置apk，安装后重新登录下 https://www.123pan.com/s/625SVv-BYxbh.html
> 
> 然后重新操作就行了。

PHP Parse error:  syntax error, unexpected token "=>" in C:\php-8.0.7\bypass.php on line 279