MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #101: openssl_decrypt() function error

zingrx, 2024-04-19:
I'm running the script on Ubuntu 23.10. PHP 8.2 is installed, and curl and openssl are enabled by default.

This is the result when I run `php -m` to check the installed/enabled modules
`[PHP Modules]
calendar
Core
ctype
curl
date
exif
FFI
fileinfo
filter
ftp
gettext
hash
iconv
json
libxml
openssl
pcntl
pcre
PDO
Phar
posix
random
readline
Reflection
session
shmop
sockets
sodium
SPL
standard
sysvmsg
sysvsem
sysvshm
tokenizer
Zend OPcache
zlib

[Zend Modules]
Zend OPcache
`

And this is the error when I run `php bypass.php`:

`[2024-04-19] [21:29:43] [INFO] - ************************************
[2024-04-19] [21:29:43] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-04-19] [21:29:43] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-04-19] [21:29:43] [INFO] - ************************************
[2024-04-19] [21:29:43] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-04-19] [21:29:43] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-04-19] [21:29:43] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-04-19] [21:29:43] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-04-19] [21:29:43] [INFO] - My Blog: https://www.neko.ink/
[2024-04-19] [21:29:43] [INFO] - ************************************
[2024-04-19] [21:29:43] [INFO] - Starting ADB server...
[2024-04-19] [21:29:43] [INFO] - Processing device a13a274a(device:manet)...
[2024-04-19] [21:29:43] [INFO] - Finding BootLoader unlock bind request...
[2024-04-19] [21:29:44] [INFO] * Now you can bind account in the developer options.
adb: invalid transport id
[2024-04-19] [21:29:44] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /media/mni/extssd/home/mni/git/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 330 and defined in /media/mni/extssd/home/mni/git/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php:228
Stack trace:
#0 /media/mni/extssd/home/mni/git/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php(330): decryptData()
#1 {main}
  thrown in /media/mni/extssd/home/mni/git/Xiaomi-HyperOS-BootLoader-Bypass/bypass.php on line 228
`

Any idea how to fix this?

hanxiang-li, 2024-04-20:
The script root directory has a php.ini file, which modifies extension_dir to the root directory of the PHP program ext
[image]

W1LLY-W0N, 2024-04-28:
Hello ! I have the same problem. I suppress « ; » in the php.ini files like this :

[...]
extension=curl
;extension=ffi
;extension=ftp
;extension=fileinfo
;extension=gd
;extension=gettext
;extension=gmp
;extension=intl
;extension=imap
;extension=mbstring
;extension=exif      ; Must be after mbstring as it depends on it
;extension=mysqli
;extension=oci8_12c  ; Use with Oracle Database 12c Instant Client
;extension=oci8_19  ; Use with Oracle Database 19 Instant Client
;extension=odbc
extension=openssl
[...]

I do not know if that's the right thing to do. If you have any idea that would be great. 
Thanks :)

xbn2002, 2024-05-05:
Look #91 pls

stablestud, 2024-05-06:
#91 does not help at all. Also the script root directory does not have a php.ini file

stablestud, 2024-05-06:
When running
`adb shell "logact *:S CloudDeviceStatus:V"`
the following output is important to note:
`DeviceStatus: content: {"code":30001,"description
[..]
Couldn't add. Please go to Mi Community to apply for authorization and try again.`
Code `30001`, based on the repos FAQ means that the hack cannot be applied to the model, bad luck for me then...

stablestud, 2024-05-06:
Anyone else desperate to unlock the bootloader of Xiaomi devices look here: https://xiaomi.eu/community/threads/unlock-by-xiaomi-community.71321/post-719423

You must do unlocking via Xiaomi Community App. Make sure you set the site location to **Global** not any other country like Germany (see screenshot of linked post)

xbn2002, 2024-05-07:
> When running `adb shell "logact *:S CloudDeviceStatus:V"` the following output is important to note:
>
> ```
> DeviceStatus: content: {"code":30001,"description
> [..]
> Couldn't add. Please go to Mi [...]

30001 means your device is not supported to bypass, just give up.

mxwmnn, 2024-05-09:
> 30001 means your device is not supported to bypass, just give up.

really or is it  maybe the xiaomi account?

have the same code in logcat. have a pad 5 pro 5g (only CN) created the account with the tablet. will try creating a new one on xiaomi site and try again :(

mxwmnn, 2024-05-09:
Don't know what was wrong here.
Have enuma and used HyperFuck. Worked. Had HyperOS 1.0.1.0

Solved for me

closed 2024-07-02 by MlgmXyysd