MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #105: No longer works

TheAirBlow, 2024-04-23:
Xiaomi changed the way logs are encrypted and the keys in the source code are no longer valid. \
The new algorithm is located at `com.android.settings.bootloader.LogEncryptor` inside `com.android.settings`.

TheAirBlow, 2024-04-23:
You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck)

Tigercat000, 2024-04-24:
> You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck)

Please help to explain how to use this Hyperfuck?thanks

TheAirBlow, 2024-04-24:
> > You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck)
> 
> Please help to explain how to use this Hyperfuck?thanks

I didn't make a good tutorial yet. If this one doesn't work - follow the instructions in the warning section.

Tigercat000, 2024-04-24:
> > > You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck)
> > 
> > 
> > Please help to explain how to use this Hyperfuck?thanks
> 
> I didn't make a good tutorial yet. If this bypass doesn't work - follow the instructions in the warning section.

Thanks for your reply , that was show this:
PS C:\HyperFuck-main> # connect your Android device before continuing
PS C:\HyperFuck-main> # make sure to enable USB debugging
PS C:\HyperFuck-main> python main.py
PS C:\HyperFuck-main> adb push Settings.apk /data/local/tmp/Settings.apk
adb server is out of date.  killing...
* daemon started successfully *
0 KB/s (134 bytes in 1.000s)
PS C:\HyperFuck-main> adb shell pm install -r -d /data/local/tmp/Settings.apk

Exception occurred while executing 'install':
java.lang.IllegalArgumentException: Error: Failed to parse APK file: /data/local/tmp/Settings.apk: Failed to parse /data/local/tmp/Settings.apk
        at com.android.server.pm.PackageManagerShellCommand.setParamsSize(PackageManagerShellCommand.java:717)
        at com.android.server.pm.PackageManagerShellCommand.doRunInstall(PackageManagerShellCommand.java:1577)
        at com.android.server.pm.PackageManagerShellCommand.runInstall(PackageManagerShellCommand.java:1543)
        at com.android.server.pm.PackageManagerShellCommand.onCommand(PackageManagerShellCommand.java:236)
        at com.android.modules.utils.BasicShellCommandHandler.exec(BasicShellCommandHandler.java:97)
        at android.os.ShellCommand.exec(ShellCommand.java:38)
        at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onShellCommand(PackageManagerService.java:6551)
        at android.os.Binder.shellCommand(Binder.java:1075)
        at android.os.Binder.onTransact(Binder.java:895)
        at android.content.pm.IPackageManager$Stub.onTransact(IPackageManager.java:4352)
        at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onTransact(PackageManagerService.java:6535)
        at android.os.Binder.execTransactInternal(Binder.java:1351)
        at android.os.Binder.execTransact(Binder.java:1282)
Caused by: java.io.IOException: Failed to load asset path /data/local/tmp/Settings.apk from fd 662
        at android.content.res.ApkAssets.nativeLoadFd(Native Method)
        at android.content.res.ApkAssets.<init>(ApkAssets.java:309)
        at android.content.res.ApkAssets.loadFromFd(ApkAssets.java:180)
        at android.content.pm.parsing.ApkLiteParseUtils.parseApkLiteInner(ApkLiteParseUtils.java:356)
        at android.content.pm.parsing.ApkLiteParseUtils.parseApkLite(ApkLiteParseUtils.java:344)
        at com.android.server.pm.PackageManagerShellCommand.setParamsSize(PackageManagerShellCommand.java:711)
        ... 12 more

TheAirBlow, 2024-04-24:
> > > > You can rollback to an older version of the settings app, [which can be found here](https://github.com/TheAirBlow/HyperFuck)
> > > 
> > > 
> > > Please help to explain how to use this Hyperfuck?thanks
> > 
> > 
> > I didn't make a good tutorial yet. If this bypass doesn't work - follow the instructions in the warning section.
> 
> Thanks for your reply , that was show this: PS C:\HyperFuck-main> # connect your Android device before continuing PS C:\HyperFuck-main> # make sure to enable USB debugging PS C:\HyperFuck-main> python main.py PS C:\HyperFuck-main> adb push Settings.apk /data/local/tmp/Settings.apk adb server is out of date. killing...
> 
>     * daemon started successfully *
>       0 KB/s (134 bytes in 1.000s)
>       PS C:\HyperFuck-main> adb shell pm install -r -d /data/local/tmp/Settings.apk
> 
> 
> Exception occurred while executing 'install': java.lang.IllegalArgumentException: Error: Failed to parse APK file: /data/local/tmp/Settings.apk: Failed to parse /data/local/tmp/Settings.apk at com.android.server.pm.PackageManagerShellCommand.setParamsSize(PackageManagerShellCommand.java:717) at com.android.server.pm.PackageManagerShellCommand.doRunInstall(PackageManagerShellCommand.java:1577) at com.android.server.pm.PackageManagerShellCommand.runInstall(PackageManagerShellCommand.java:1543) at com.android.server.pm.PackageManagerShellCommand.onCommand(PackageManagerShellCommand.java:236) at com.android.modules.utils.BasicShellCommandHandler.exec(BasicShellCommandHandler.java:97) at android.os.ShellCommand.exec(ShellCommand.java:38) at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onShellCommand(PackageManagerService.java:6551) at android.os.Binder.shellCommand(Binder.java:1075) at android.os.Binder.onTransact(Binder.java:895) at android.content.pm.IPackageManager$Stub.onTransact(IPackageManager.java:4352) at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onTransact(PackageManagerService.java:6535) at android.os.Binder.execTransactInternal(Binder.java:1351) at android.os.Binder.execTransact(Binder.java:1282) Caused by: java.io.IOException: Failed to load asset path /data/local/tmp/Settings.apk from fd 662 at android.content.res.ApkAssets.nativeLoadFd(Native Method) at android.content.res.ApkAssets.(ApkAssets.java:309) at android.content.res.ApkAssets.loadFromFd(ApkAssets.java:180) at android.content.pm.parsing.ApkLiteParseUtils.parseApkLiteInner(ApkLiteParseUtils.java:356) at android.content.pm.parsing.ApkLiteParseUtils.parseApkLite(ApkLiteParseUtils.java:344) at com.android.server.pm.PackageManagerShellCommand.setParamsSize(PackageManagerShellCommand.java:711) ... 12 more

You cloned the repo without LFS files. Install Git LFS or use the ZIP download feature.

Tigercat000, 2024-04-24:
> You cloned the repo without LFS files. Install Git LFS or use the ZIP download feature.

Try to  Install Git LFS but show
could not find Git can not register git lfs
How can install it ? Thanks

TheAirBlow, 2024-04-24:
> Try to Install Git LFS but show could not find Git can not register git lfs How can install it ? Thanks

Install git lol

or just download the APK directly from github by clicking Raw.

Tigercat000, 2024-04-24:
> Install git lol
>
> or just download the APK directly from github by clicking Raw.

Install git already , I download HyperFuck zip ,after unzip file but can't install setting.apk,thanks

conflagration27, 2024-04-24:
Will this work on devices that have HyperOS CN installed from the factory? Such as Xiaomi 14 series, Redmi K70 etc
ty

TheAirBlow, 2024-04-24:
> Got error:
> 
> ```
> $ python3 main.py 
> Found Xiaomi Xiaomi (board shennong, codename shennong)
> Is this correct? (y/N): y
> Open MI Unlock Status and request permission
> Disabling mobile internet, taking over!
> Decrypting arguments from logs
> Decrypting headers from logs
> Impersonating MIUI 14 to bypass requirements
> Sending forged account binding request
> Device forced to verify through Mi Community (30001)
> ```
> 
> This is the end?(

Yup, can't do anything.

Tigercat000, 2024-04-24:
Is too sad end of K70 series ,but thanks all of your hard work ,appreciate.

TheAirBlow, 2024-04-25:
@Tigercat000 Try [HyperSploit](https://github.com/TheAirBlow/HyperSploit/releases/), it's quite a bit more user-friendly with the downgrading shenanigans.

Tigercat000, 2024-04-25:
> @Tigercat000 Try [HyperSploit](https://github.com/TheAirBlow/HyperSploit/releases/), it's quite a bit more user-friendly with the downgrading shenanigans.

Thanks buddy , appreciate

TheAirBlow, 2024-04-25:
I'm going to close this as the author is clearly aware of the issue judging by their twitter

closed 2024-04-25 by TheAirBlow

spiderfix101, 2024-05-05:
I need to unlock the bootloader on Redmi Note 13 Pro 5G China version. Will this work on this device?

TheAirBlow, 2024-05-05:
> I need to unlock the bootloader on Redmi Note 13 Pro 5G China version. Will this work on this device?

Chinese devices are basically fucked.