MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #106: PHP Fatal error:  Uncaught Error: Call to undefined function curl_init() after Sending POST request

drus519, 2024-04-23:
Have next log when trying to bind account in developer settings:
```
[2024-04-24] [01:03:51] [INFO] - ************************************
[2024-04-24] [01:03:51] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-04-24] [01:03:51] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-04-24] [01:03:51] [INFO] - ************************************
[2024-04-24] [01:03:51] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-04-24] [01:03:51] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-04-24] [01:03:51] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-04-24] [01:03:51] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-04-24] [01:03:51] [INFO] - My Blog: https://www.neko.ink/
[2024-04-24] [01:03:51] [INFO] - ************************************
[2024-04-24] [01:03:51] [INFO] - Starting ADB server...
[2024-04-24] [01:03:51] [INFO] - Processing device 9b04aa0b(device:shennong)...
[2024-04-24] [01:03:51] [INFO] - Finding BootLoader unlock bind request...
[2024-04-24] [01:03:51] [INFO] * Now you can bind account in the developer options.
[2024-04-24] [01:04:02] [INFO] - Account bind request found! Let's block it.
[2024-04-24] [01:04:02] [INFO] - Refactoring parameters...
[2024-04-24] [01:04:02] [INFO] - Sending POST request...
PHP Fatal error:  Uncaught Error: Call to undefined function curl_init() in /home/drus/MiUnlockTool/php/bypass.php:158
Stack trace:
#0 /home/drus/MiUnlockTool/php/bypass.php(199): http('https://unlock....', 'POST', 'data=%7B%22clou...', Array, true)
#1 /home/drus/MiUnlockTool/php/bypass.php(353): postApi('unlock/applyBin...', Array, Array, true)
#2 {main}
  thrown in /home/drus/MiUnlockTool/php/bypass.php on line 158

Fatal error: Uncaught Error: Call to undefined function curl_init() in /home/drus/MiUnlockTool/php/bypass.php:158
Stack trace:
#0 /home/drus/MiUnlockTool/php/bypass.php(199): http('https://unlock....', 'POST', 'data=%7B%22clou...', Array, true)
#1 /home/drus/MiUnlockTool/php/bypass.php(353): postApi('unlock/applyBin...', Array, Array, true)
#2 {main}
  thrown in /home/drus/MiUnlockTool/php/bypass.php on line 158
``` 

Tried to downgrade "Settings" app like in [comment](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/29#issuecomment-1900298533) by [APK](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip) but still get the errors mentioned above

p-hao, 2024-04-27:
1. Make sure there is a `php.in` file in your php install location.
2. Create a php file with content:
```
<?php
 phpinfo();
?>
```
3. Run the php file and then copy the result to your text edit
4. Search `Loaded Configuration File` to Check if  any config document( php.ini) is loaded. The correct result may like `Loaded Configuration File => C:\php-8.3.6-Win32-vs16-x86\php.ini`
5. Search `openssl`, the correct result may like `OpenSSL support => enabled` , if its disabled, delete `;` before `extension=openssl` in `php.ini`file

popkter, 2024-06-29:
> 2. Create a php file with content:
>
> ```
> <?php
>  phpinfo();
> ?>
> ```
>
> 3. Run the php file and then copy the result to your text edit
> 4. Search `Loaded Configuration File` to Check if  any config [...]

I sware openssl has been setup correctly，but this script do not work on my Windows11 PC，I install php env by XAMPP

popkter, 2024-06-30:
oh god i find something wrong .if use cmd script ，it cannot find php extenstions even though I has add config to system path。if cmd doesnot work, try run  php bypass.php directly.

closed 2024-07-02 by MlgmXyysd