MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #108: 小米真够了

hanxiang-li, 2024-04-26:
我的小米14pro无法解锁，然后我的小米 6s pro 直接解锁成功。
[image]
不知道为啥用Xiaomi-HyperOS-BootLoader-Bypass 一直参数错误，换了好几个php版本，

hanxiang-li, 2024-04-26:
[image]

gsmlm, 2024-04-27:
哈哈，有python版的还有c#版的，都参考的这个php版的。参数错误基本都是 rsa加密版本，替换settings.apk，有的可以有的不行。6spro?也澎湃吗？miui不是可以直接解吗？

p-hao, 2024-04-27:
> 哈哈，有python版的还有c#版的，都参考的这个php版的。参数错误基本都是 rsa加密版本，替换settings.apk，有的可以有的不行。6spro?也澎湃吗？miui不是可以直接解吗？

我也是解密那一段代码返回false, 是需要替换手机的设置程序吗，方便指个路不

gsmlm, 2024-04-27:
> > 哈哈，有python版的还有c#版的，都参考的这个php版的。参数错误基本都是 rsa加密版本，替换settings.apk，有的可以有的不行。6spro?也澎湃吗？miui不是可以直接解吗？
> 
> 我也是解密那一段代码返回false, 是需要替换手机的设置程序吗，方便指个路不

https://github.com/TheAirBlow/HyperFuck

klcb2010, 2024-04-27:
> @p-hao
> 
> > 我也是解密那一段代码返回false, 是需要替换手机的设置程序吗，方便指个路不
> 
> https://github.com/TheAirBlow/HyperFuck 试试这个

Xiaomi quickly patched this method out.

p-hao, 2024-04-27:
> @p-hao
> 
> > 我也是解密那一段代码返回false, 是需要替换手机的设置程序吗，方便指个路不
> 
> https://github.com/TheAirBlow/HyperFuck 试试这个

按照 issue #82 替换 `settings.app`就可以了

hanxiang-li, 2024-04-27:
> > @p-hao
> > > 我也是解密那一段代码返回false, 是需要替换手机的设置程序吗，方便指个路不
> > 
> > 
> > https://github.com/TheAirBlow/HyperFuck 试试这个
> 
> 按照 issue #82 替换 `settings.app`就可以了

小米14pro不行

xbn2002, 2024-05-04:
> > > @p-hao
> > > > 我也是解密那一段代码返回false, 是需要替换手机的设置程序吗，方便指个路不
> > > 
> > > 
> > > https://github.com/TheAirBlow/HyperFuck 试试这个
> > 
> > 
> > 按照 issue #82 替换 `settings.app`就可以了
> 
> 小米14pro不行

出厂澎湃无法绕过

xbn2002, 2024-05-05:
> 我的小米14pro无法解锁，然后我的小米 6s pro 直接解锁成功。 [...]

收到货了很多遍了，出厂澎湃的机子没办法绕过