MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #115: Mi 11 Ultra - Request parameter error (10000)

dinhtruyen93, 2024-05-03:
[image]
help me

chien132, 2024-05-05:
> ![1](https://private-user-images.githubusercontent.com/121920055/327772561-58eb4c5e-c58d-45ac-96c0-e8cb83f82fed.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ4ODUwMjYsIm5iZiI6MTcxNDg4NDcyNiwicGF0aCI6Ii8xMjE5MjAwNTUvMzI3NzcyNTYxLTU4ZWI0YzVlLWM1OGQtNDVhYy05NmMwLWU4Y2I4M2Y4MmZlZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNVQwNDUyMDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lN2U4MjA0N2E0NzUxMTBkODEwYzBlOGIwMWY2MjQ2NjljN2IyMmVjMThlZTFmMTFjYTlkNjlkMDc5NWYyMDBiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wQ85VhpsuYYFjA9Wia6cmbTM-bgH1UDgiNiKYuiVxWE) The problem has been resolved successfully Use the installation file at the link below
> 
> `https://drive.google.com/file/d/1I8yXLy6vSsk2uxp--rB8cal0K73gxQ_j/view?usp=drive_link`

I'm encounter the same issue on K60Ultra, could you tell me where did you get this file, is this the same file with https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip ?

dinhtruyen93, 2024-05-05:
> I'm encounter the same issue on K60Ultra, could you tell me where did you get this file, is this the same file with [...]

- [ ] https://www.123pan.com/s/625SVv-BYxbh.html

chien132, 2024-05-05:
> * [x]  https://www.123pan.com/s/625SVv-BYxbh.html

Đã thành công, đa tạ anh trai 😁

closed 2024-05-05 by dinhtruyen93