MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #117: 进入开发者选项闪退

Miyoru, 2024-05-04:
安装了Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip里的Settings.apk后，进入开发者选项闪退怎么办？

chien132, 2024-05-05:
Same isssue on Redmi K60 Ultra running 1.0.12.0UMLCNXM, cannot go to Developer options after install [Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip]

xbn2002, 2024-05-05:
答案是找一个能用的settings.apk，请查看其他issue或者在各大论坛上搜索
The answer is to find a working settings.apk. Please check other issues or search on major forums.

Keritial, 2024-06-02:
The Settings apk provided in the Release works on Redmi Note 11T Pro OS1.0.1.0.ULOCNXM but not on OS1.0.2.0.ULOCNXM. I guess that Xiaomi has changed something related to Settings. By the way, not only Developer Options but also Bluetooth settings crash too.

phpadminer, 2025-01-18:
我是 redmi K60 U , 我使用了 https://xmomo.lanzoum.com/i5VRs1vzpfxg 这个里面的Setting,之后在应用管理下找到设置，然后清空所有数据就好了。

> Same isssue on Redmi K60 Ultra running 1.0.12.0UMLCNXM, cannot go to Developer options after install [Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip]