MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #123: PHP Fatal error:  Uncaught TypeError: decryptData():

Tuiye, 2024-05-17:
PHP Fatal error:  Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given, called in /xxx/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php on line 330 and defined in /xxx/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php:228
Stack trace:
#0 /xxx/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php(330): decryptData(NULL)
#1 {main}
  thrown in /xxx/Xiaomi-HyperOS-BootLoader-Bypass-master/bypass.php on line 228

Tuiye, 2024-05-17:
MacOS

pulpoff, 2024-06-03:
same error on ubuntu as well

popkter, 2024-06-29:
same error on Win11, install php env by xampp

closed 2024-07-02 by MlgmXyysd

mhmdali711, 2024-07-04:
i have the same problem with my redmi turbo 3 chinese rom

cnjsyyb, 2024-09-10:
根本不能用

SergeTDU, 2025-03-04:
Has anyone figure it out? getting the same error