MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #125: 解决 CloudDeviceStatus 数据中嵌入标签预期导致的 openssl_decrypt 返回 null 问题

aljbri, 2024-05-23:
### 问题

`openssl_decrypt` 函数在第230行返回了一个 `null` 值，导致以下错误：
```
PHP Fatal error: Uncaught TypeError: decryptData(): Argument #1 ($data) must be of type string, null given
```

#### 分析
经过调查发现，第311行返回的值包含了一个模式，其中数据以 `#&^` 开头，并以 `^&#` 结尾。这些序列似乎充当了表示实际数据开始和结束的标签。此外，我还发现 `args` 和 `headers` 中的前346个字符是相同的。

#### 数据示例
以下是获取 `CloudDeviceStatus` 信息时提取的数据示例：
```CloudDeviceStatus
05-23 14:53:31.867 27502 27523 D CloudDeviceStatus: stateCode: 200

05-23 14:53:31.867 27502 27523 D CloudDeviceStatus: content: {"result":"ok","retriable":false,"code":0,"data":{"nonce":"AgUAHHJ........NT0ybdeg"},"description":"成功","ts":1716447211661}

05-23 14:53:32.025 27502 27523 I CloudDeviceStatus: args: #&^iraiCGggO0My0pKgKJlvKBIHDNPxds+............+f435KDJGw2cdVh32g==^&#

05-23 14:53:32.025 27502 27523 I CloudDeviceStatus: headers: #&^iraiCGggO0My0pKgKJlvKBIHDNPxds+............+6PhM2LWzPO6QcLtpqNw4hj/^&#
```

这些标签可能导致了解密过程失败，导致返回 `null` 值。为了解决这个问题，应在将数据传递给 `openssl_decrypt` 函数之前移除这些标签。

Tigercat000, 2024-05-27:
你有修改後更好的分享?

MrTang23, 2024-06-21:
CloudDeviceStatus获取不到任何内容，是因为hyperos的原因吗？