MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #131: Binding failed with error code 30003

Millionerovic, 2024-06-11:
[2024-06-11] [05:48:12] [INFO] - Starting ADB server...
[2024-06-11] [05:48:13] [INFO] - Processing device 16719b0a(7)...
[2024-06-11] [05:48:13] [INFO] - Finding BootLoader unlock bind request...
[2024-06-11] [05:48:14] [INFO] * Now you can bind account in the developer options.
[2024-06-11] [05:48:23] [INFO] - Account bind request found! Let's block it.
[2024-06-11] [05:48:23] [INFO] - Refactoring parameters...
[2024-06-11] [05:48:23] [INFO] - Sending POST request...
[2024-06-11] [05:48:26] [INFO] - The system is being upgraded. Please try again later. (30003)

MergenXu, 2024-06-11:
我也是这样，红米note 13 5G

MergenXu, 2024-06-11:
me too,Redmi note 13 5G

ZaryabKhan, 2024-06-12:
Bind my account successfully yesterday without any issue...I am not 100% sure but it looks like the issue is from the Xiaomi's server side.

closed 2024-07-02 by MlgmXyysd

malikoff05, 2024-10-31:
[image]

malikoff05, 2024-10-31:
Please help, poco x6 5g xyperos 1