MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #135: Xiaomi 12sultra | $args解码后$data为空 | Request parameter error (10000)

CedarXuesong, 2024-06-21:
[image]
中间那段是自己输出的$args
经过decryptData后什么都没得到

CedarXuesong, 2024-06-21:
Hyperos版本 1.0.5.0.ULACNXM

ElYaiko, 2024-06-23:
Same error

tabs, 2024-06-25:
Same error

thanhthuongpham, 2024-06-25:
I also got the same error with xiaomi 14 ultra

hubing1994, 2024-06-29:
我是小米13，一样的报错

hubing1994, 2024-06-29:
> ![屏幕截图 2024-06-22 015311](https://private-user-images.githubusercontent.com/173494518/341849785-3db7371e-ac63-4a6e-873e-5894f4467b36.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk2NjcyMDYsIm5iZiI6MTcxOTY2NjkwNiwicGF0aCI6Ii8xNzM0OTQ1MTgvMzQxODQ5Nzg1LTNkYjczNzFlLWFjNjMtNGE2ZS04NzNlLTU4OTRmNDQ2N2IzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyOVQxMzE1MDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OWJmNzk3OGQwZTBkYTczMmUzZDRiNzkyZTMxNjg3ZTVmOGU1ZDk2Yzg1M2E0Nzc1ZmRmYTExN2I1MTNlYWZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3nQfTAuXe9V6QezHQADAF5Skzj2mQCgedDN04iXQsyw) 中间那段是自己输出的$args 经过decryptData后什么都没得到

我这边验证了以下，和你一样的问题，澎湃系统的版本是开发版，1.0.24.6.17.DEV

BlueGradientHorizon, 2024-08-03:
@MlgmXyysd, why did you close the issue? The problem persists and wasn't solved. Please reopen issue or clarify why it was closed. Thanks.