MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #137: OpenSSL has been setup correctly，but throw exception

popkter, 2024-06-29:
this is my phpinfo
[image]

when  i run bypass.cmd and click bind account on device，throw exception：
> 
[image]

I am not good at php, any one can help me? thank u very much!

popkter, 2024-06-30:
oh god i find something wrong .if use cmd script ，it cannot find php extenstions even though I has add config to system path。if   cmd doesnot work, try run ` php bypass.php` directly.

closed 2024-06-30 by popkter

mhmdali711, 2024-07-04:
> oh god i find something wrong .if use cmd script ，it cannot find php extenstions even though I has add config to system path。if cmd doesnot work, try run ` php bypass.php` directly.

please any help with this problem

Oyama-Oyama, 2024-08-26:
same error
any help？

popkter, 2024-08-26:
> same error any help？

i directly run bypass.php to fix this error

VitmaSense, 2025-09-16:
@popkter  Can you please explain how to get rid of the error?