MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #140: hello , i have this problem facing me while use php to unlock , please need help

mhmdali711, 2024-07-04:
PHP Warning:  PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (The specified module could not be found), C:\php\ext\php_curl.dll (The specified module could not be found)) in Unknown on line 0

Warning: PHP Startup: Unable to load dynamic library 'curl' (tried: C:\php\ext\curl (The specified module could not be found), C:\php\ext\php_curl.dll (The specified module could not be found)) in Unknown on line 0
PHP Warning:  PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (The specified module could not be found), C:\php\ext\php_openssl.dll (The specified module could not be found)) in Unknown on line 0

Warning: PHP Startup: Unable to load dynamic library 'openssl' (tried: C:\php\ext\openssl (The specified module could not be found), C:\php\ext\php_openssl.dll (The specified module could not be found)) in Unknown on line 0
[2024-07-04] [11:49:50] [INFO] - ************************************
[2024-07-04] [11:49:50] [INFO] - * Xiaomi HyperOS BootLoader Bypass *
[2024-07-04] [11:49:50] [INFO] - * By NekoYuzu          Version 1.0 *
[2024-07-04] [11:49:50] [INFO] - ************************************
[2024-07-04] [11:49:50] [INFO] - GitHub: https://github.com/MlgmXyysd
[2024-07-04] [11:49:50] [INFO] - XDA: https://xdaforums.com/m/mlgmxyysd.8430637
[2024-07-04] [11:49:50] [INFO] - X (Twitter): https://x.com/realMlgmXyysd
[2024-07-04] [11:49:50] [INFO] - PayPal: https://paypal.me/MlgmXyysd
[2024-07-04] [11:49:50] [INFO] - My Blog: https://www.neko.ink/
[2024-07-04] [11:49:50] [INFO] - ************************************
[2024-07-04] [11:49:50] [INFO] - Starting ADB server...
[2024-07-04] [11:49:51] [INFO] - Processing device 7db16272(3)...
[2024-07-04] [11:49:52] [INFO] - Finding BootLoader unlock bind request...
[2024-07-04] [11:49:52] [INFO] * Now you can bind account in the developer options.
[2024-07-04] [11:50:01] [INFO] - Account bind request found! Let's block it.
[2024-07-04] [11:50:01] [INFO] - Refactoring parameters...
PHP Fatal error:  Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\lenovo\Desktop\New folder (2)\php\bypass.php:230
Stack trace:
#0 C:\Users\lenovo\Desktop\New folder (2)\php\bypass.php(330): decryptData('#&^DsP2rhpJdKJo...')
#1 {main}
  thrown in C:\Users\lenovo\Desktop\New folder (2)\php\bypass.php on line 230

Fatal error: Uncaught Error: Call to undefined function openssl_decrypt() in C:\Users\lenovo\Desktop\New folder (2)\php\bypass.php:230
Stack trace:
#0 C:\Users\lenovo\Desktop\New folder (2)\php\bypass.php(330): decryptData('#&^DsP2rhpJdKJo...')
#1 {main}
  thrown in C:\Users\lenovo\Desktop\New folder (2)\php\bypass.php on line 230

SquadCraftFTW, 2024-07-09:
same issue on xiaomi pad 5 EEA

SquadCraftFTW, 2024-07-19:
Fixed it by looking at issue #3 
Google how to enable curl and openssl extension in php, and make sure the directory of the "ext" folder is like in the picture.