MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #143: Error on some post request

baguetteFan, 2024-07-15:
```
[2024-07-01] [12:48:01] [INFO] - Finding BootLoader unlock bind request...
[2024-07-01] [12:48:01] [INFO] * Now you can bind account in the developer options.
[2024-07-01] [12:48:04] [INFO] - Account bind request found! Let's block it.
[2024-07-01] [12:48:04] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in D:\miui\Bypass-1.0-fix-universal\bypass.php on line 333

Warning: Trying to access array offset on null in D:\miui\Bypass-1.0-fix-universal\bypass.php on line 333
PHP Deprecated:  str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in D:\miui\Bypass-1.0-fix-universal\bypass.php on line 333

Deprecated: str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in D:\miui\Bypass-1.0-fix-universal\bypass.php on line 333
[2024-07-01] [12:48:04] [INFO] - Sending POST request...
[2024-07-01] [12:48:06] [INFO] ! invalid request parameters
```

Thanks i really need to haxor into my phone

baguetteFan, 2024-07-16:
was on redmi note 12 nfc

baguetteFan, 2024-07-18:
@IANthji have you had bad cell phone data thing? I barely had any so maybe this was the issue?

PeterTerpe, 2024-11-03:
same issue here, redmi k50

omrdncr, 2025-01-11:
Same issue Xiaomi 14 ultra Titanium