MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #146: Script gets stuck at "Now you can bind account in the developer options."

Arna-Maity, 2024-07-21:
Once I run the script, it gets stuck at this state. Even after 10 mins there is no progress.

[image]

Adlexer, 2024-07-25:
Check if your "Select Log Level" option is "Off" and set it to "Info"

Arna-Maity, 2024-07-27:
@Adlexer 
Hi, Thanks for your response. I already got the bootloader unlock permission from the Mi Community App the official way.
If required, you can close this issue.

kabuda2077, 2024-09-15:
> 检查“选择日志级别”选项是否为“关闭”并将其设置为“信息” ![IMG_20240725_105558.jpg](https://private-user-images.githubusercontent.com/40536337/351941886-2e54f0b8-acb9-4447-a6c3-448a90cb78f8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzNjcxODIsIm5iZiI6MTcyNjM2Njg4MiwicGF0aCI6Ii80MDUzNjMzNy8zNTE5NDE4ODYtMmU1NGYwYjgtYWNiOS00NDQ3LWE2YzMtNDQ4YTkwY2I3OGY4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE1VDAyMjEyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI5MGJiYzc4YjM4Mjc3MzdjNzU1MDdkZTNjYjU3MzRkN2JjMjU3ZDVmMTg3OTViYTI3ZDJhNWZjMzNiNmUwYWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZB-zJSjoCncZceSSJ7fE2-kb-XDRJr9b7IxmdZyLKxA)

I also encountered the same problem. The phone settings flashed but the account binding interface did not pop up, and it did not work according to your settings.

luigiradano, 2024-10-20:
Thanks @Adlexer your solution seems to have solved the problem form me!

zhiyuanisme, 2024-11-22:
i meet the same problem ,but I can't solve it ,what issue that 
 you solve it