MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass #152: Poco F5 Pro/K60 not working

Sm0kedPaprika, 2024-07-27:
[2024-07-27] [11:03:12] [INFO] - Starting ADB server...
[2024-07-27] [11:03:18] [INFO] - Processing device 50d07ca1(1)...
[2024-07-27] [11:03:18] [INFO] - Finding BootLoader unlock bind request...
[2024-07-27] [11:03:19] [INFO] * Now you can bind account in the developer options.
[2024-07-27] [11:03:28] [INFO] - Account bind request found! Let's block it.
[2024-07-27] [11:03:28] [INFO] - Refactoring parameters...
PHP Warning:  Trying to access array offset on null in C:\php\bypass.php on line 333

Warning: Trying to access array offset on null in C:\php\bypass.php on line 333
PHP Deprecated:  str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in C:\php\bypass.php on line 333

Deprecated: str_replace(): Passing null to parameter #3 ($subject) of type array|string is deprecated in C:\php\bypass.php on line 333
[2024-07-27] [11:03:28] [INFO] - Sending POST request...
[2024-07-27] [11:03:29] [INFO] - Request parameter error (10000)

Sm0kedPaprika, 2024-07-27:
I applied the fix and now I get:

[2024-07-27] [11:36:04] [INFO] - Starting ADB server...
[2024-07-27] [11:36:04] [INFO] - Waiting for device connection...
[2024-07-27] [11:36:06] [INFO] - Processing device 50d07ca1(1)...
[2024-07-27] [11:36:06] [INFO] - Finding BootLoader unlock bind request...
[2024-07-27] [11:36:06] [INFO] * Now you can bind account in the developer options.
[2024-07-27] [11:36:12] [INFO] - Account bind request found! Let's block it.
[2024-07-27] [11:36:12] [INFO] - Refactoring parameters...
[2024-07-27] [11:36:12] [INFO] - Sending POST request...
[2024-07-27] [11:36:16] [INFO] - Target account: 8189239777
[2024-07-27] [11:36:16] [INFO] - Account bound successfully, wait time can be viewed in the unlock tool.

But the account is not bound anyway.

Sm0kedPaprika, 2024-07-27:
consider trying [https://github.com/TheAirBlow/HyperSploit/](url)

Arna-Maity, 2024-07-27:
I guess you are try to view your account binding status from the Mi Unlock Status page of your phone. Based on the following comment that would not work. You need to use the Mi Unlock Tool to check for your wait duration now.

https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/issues/76#issuecomment-1988667038

Sm0kedPaprika, 2024-08-05:
[image]

closed 2024-08-05 by Sm0kedPaprika